mjg59/python-broadlink #503: MCB1 device: None None (0xa56a) at ......

amosrohe, 2021-01-11:
hi
trying to work with the MCB1 and it looks like it recognizes it but not the commands

dev
<device: None None (0xa56a) at 192.168.0.184:80 | a0:44:b0:2b:ee:d7 | 42346 | Unlocked>

dev.auth()
True
dev.get_fwversion()
57052
dev.get_state()
Traceback (most recent call last):
File "", line 1, in
AttributeError: 'device' object has no attribute 'get_state'

same for power and other instructions
the line 0xa56a: (sp4, "MCB1", "Broadlink"), exists in the __init__py file

tnx

KTibow, 2021-01-11:
Is it unlocked?

amosrohe, 2021-01-11:
yes 
device: None None (0xa56a) at 192.168.0.184:80 | a0:44:b0:2b:ee:d7 | 42346 | **Unlocked**>

KTibow, 2021-01-11:
If you reset it, and pair it without the broadlink app, do you still hace the problem?

amosrohe, 2021-01-11:
when connecting using the broadlink app it is locked and get an Authorization issue

amosrohe, 2021-01-11:
using the broadlink app to set on and off works .. 
when pairing using the app the python et an Authorization issue

amosrohe, 2021-01-11:
reset it, and pair it without the broadlink app, 
I get this
device: None None (0xa56a) at 192.168.0.184:80 | a0:44:b0:2b:ee:d7 | 42346 | Unlocked>

felipediel, 2021-01-11:
Hi @amosrohe. If that line existed, you would be getting an sp4 object as response.

Your response:
```
<device: None None (0xa56a) at 192.168.0.184:80 | a0:44:b0:2b:ee:d7 | 42346 | Unlocked>
```
Should be:
```
<sp4: None None (0xa56a) at 192.168.0.184:80 | a0:44:b0:2b:ee:d7 | 42346 | Unlocked>
```

So you don't have the latest version in your environment. To install it:
```
pip3 install git+https://github.com/mjg59/python-broadlink.git@master --force-reinstall --no-deps
```

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel