mjg59/python-broadlink #512: SCB1E (0x5115)

RenierM26, 2021-01-15:
Hi There,

Could you please add support for scb1e with code 0x5115?

It's a broadlink bestcon power metering wifi switch with 16a max current.

I'm more than willing to test or provide any info required.

Thank you for all the work on this api!

RenierM26, 2021-01-17:
#514

RenierM26, 2021-01-19:
After adding

0x5115: (sp4b, "SCB1E", "Broadlink"),

to __init__.py

C:\Users\renier\AppData\Roaming\Python\Python39\site-packages\broadlink>python
Python 3.9.1 (tags/v3.9.1:1e5d33e, Dec  7 2020, 17:08:21) [MSC v.1927 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink as blk
>>> dev=blk.hello("192.168.1.238")
>>> dev.auth()
True
>>> dev.get_state()
{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'current': 0.0, 'volt': 230.6, 'power': 0.0, 'totalconsum': 265.37, 'overload': 0.0}
>>> dev.get_state()
{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'current': 0.0, 'volt': 230.6, 'power': 0.0, 'totalconsum': 265.37, 'overload': 0.0}
>>> dev.get_state()
{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'current': 0.0, 'volt': 230.6, 'power': 0.0, 'totalconsum': 265.37, 'overload': 0.0}
>>> dev.get_state()
{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'current': 0.0, 'volt': 230.6, 'power': 0.0, 'totalconsum': 265.37, 'overload': 0.0}
>>> dev.get_state()
{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'current': 0.0, 'volt': 230.7, 'power': 0.0, 'totalconsum': 265.37, 'overload': 0.0}
>>> dev
<sp4b: Broadlink SCB1E (0x5115) at 192.168.1.238:80 | c8:f7:42:f8:d8:a6 | SCB1E | Unlocked>
>>>

RenierM26, 2021-01-19:
100%

KTibow, 2021-01-23:
Could this be closed then?

RenierM26, 2021-01-23:
Hi there, 

Resolved. Thank you!

closed 2021-01-23 by RenierM26