mjg59/python-broadlink #513: RM4 Pro

PregLizZz, 2021-01-15:
Hi,
I'm trying to learn some RF codes and it's failing on the step find_rf_packet(). IR Codes work fine.
On the RF codes:
found = devices[0].check_frequency()
This step still works and returns true.
I then try 
found = devices[0].find_rf_packet()
Which exits in an error:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/broadlink-0.16.0-py3.7.egg/broadlink/remote.py", line 53, in find_rf_packet
    return resp[0] == 1
IndexError: index out of range
What's the problem here?
Thanks and best regards

msbc42, 2021-01-16:
I'm seeing the same issue.
```
$ ./broadlink_cli --device @rmpro.device --learnfile fan.office --rfscanlearn
Learning RF Frequency, press and hold the button to learn...
Found RF Frequency - 1 of 2!
You can now let go of the button
Press enter to continue...
To complete learning, single press the button you want to learn
Traceback (most recent call last):
  File "./broadlink_cli", line 219, in <module>
    dev.find_rf_packet()
  File "/home/mark/.local/lib/python3.8/site-packages/broadlink/remote.py", line 52, in find_rf_packet
    return resp[0] == 1
IndexError: index out of range

```
When I press 'enter to continue' as instructed the error occurs - I don't have opportunity to respond to the "To complete learning, single press the button you want to learn" msg.

PregLizZz, 2021-01-16:
Exactly same behaviour like msbc42 when I try with --rfscanlearn.
My device is a 0x61a2

KTibow, 2021-01-18:
Looks like it's returning an empty list response. A more detailed title for this issue would be "RM4 Pro returning empty list response". @felipediel any ideas why this is happening?

felipediel, 2021-01-19:
Hi. I made a possible [fix](https://github.com/mjg59/python-broadlink/pull/516). Would you mind helping me test it? I don't have the device.
Something tells me that we have to fix rm4.check_frequency() too, I will do it depending on the results.

PregLizZz, 2021-01-19:
I really appreciate your help :) It's working. Thanks a lot.

felipediel, 2021-01-19:
Thank you for testing and reporting back!

felipediel, 2021-02-17:
Fixed. Thank you!

closed 2021-02-17 by felipediel