mjg59/python-broadlink #519: Cannot connect to light bulb - readme needs to be updated?

kylemd, 2021-01-22:
I am having issues connecting to my new light bulbs. They are Broadlink LB27 R1s, and I believe they do have Fastcon enabled.

Connecting to the bulbs SSID in AP mode and setting the WiFi settings works as intended.

I have followed the instructions on the readme but took me a while to realise the IP address needs to be encapsulated in quotes - this needs to be fixed for the newbies like myself 😂

I have received the following error trying to connect to one of the bulbs though - is there anything I can do to provide more information?

(All bulbs have had the device lock turned off from the Broadlink app)

[image]
[image]

felipediel, 2021-01-25:
Thank you @kylemd! I fixed the documentation in the dev branch, it will be merged soon.
You exposed your password. Now you need to change it.

kylemd, 2021-01-26:
> Thank you @kylemd! I fixed the documentation in the dev branch, it will be merged soon.
> You exposed your password. Now you need to change it.

Thanks @felipediel. Didn't even realise I had the password showing, whoops.

closed 2021-02-17 by felipediel