mjg59/python-broadlink #522: Add support for 0xa4f4 - Broadlink LB27 R1

kylemd, 2021-01-23:
As per title.

__Trying to get device state with no edits results in__

```
>>> import broadlink
>>> broadlink.setup('SSID','PASSWORD',3)
>>> devices=broadlink.discover(timeout=20)
>>> devices                                             
[<device: None None (0xa4f4) at 192.168.1.137:80 | a0:43:b0:1b:88:e4 | Smart Bulb | Locked>, 
<device: None None (0xa4f4) at 192.168.1.177:80 | a0:43:b0:1b:87:0f | Smart Bulb | Unlocked>]
>>> devices[0].auth()
True                                                    
>>> devices[0].set_lock(False)                          
>>> devices                                             
[<device: None None (0xa4f4) at 192.168.1.137:80 | a0:43:b0:1b:88:e4 | Smart Bulb | Unlocked>, 
<device: None None (0xa4f4) at 192.168.1.177:80 | a0:43:b0:1b:87:0f | Smart Bulb | Unlocked>]
>>> devices[0].get_type()
'Unknown'                                               
>>> devices[0].get_fwversion()
57218
>>> devices[0].get_state()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'device' object has no attribute 'get_state'
>>> devices[0].set_type("light")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'device' object has no attribute 'set_type'
```

__Trying to add 0xa4f4 to init.py resulted in the following:__

```
>>> import broadlink                                    
>>> devices = broadlink.discover(timeout=20)            
>>> devices                                             
[<lb1: Broadlink LB1 (0xa4f4) at 192.168.1.177:80 | a0:43:b0:1b:87:0f | Smart Bulb | Unlocked>, 
<lb1: Broadlink LB1 (0xa4f4) at 192.168.1.137:80 | a0:43:b0:1b:88:e4 | Smart Bulb | Locked>]
>>> devices[0].auth()                                   
True
>>> devices[0].get_state()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/data/com.termux/files/home/python-broadlink/broadlink/light.py", line 73, in get_state
    self.send_command(cmd)
  File "/data/data/com.termux/files/home/python-broadlink/broadlink/light.py", line 50, in send_command
    responseLength = int(payload[0x0A]) | (int(payload[0x0B]) << 8)
IndexError: index out of range
```
Is there any other logs I can pull to help make this work?

Just in case it helps I had some spare ones so I tore one down (destructively - for the greater good!)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kylemd, 2021-01-23:
If this helps, I tried dev branch too:

```
>>> devices[1].get_state()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/data/com.termux/files/home/python-broadlink/python-broadlink/broadlink/light.py", line 33, in get_state
    check_error(response[0x22:0x24])
  File "/data/data/com.termux/files/home/python-broadlink/python-broadlink/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.StorageError: [Errno -5] The device storage is full
```

KTibow, 2021-01-23:
Huh. @felipediel any ideas what's going on here?

kylemd, 2021-01-24:
> Huh. @felipediel any ideas what's going on here?

Any idea how I can get the payload to print so I can post here for further debug purposes?

KTibow, 2021-01-24:
Add a `print(response)` before the `check_error` call.

kylemd, 2021-01-24:
> Add a `print(response)` before the `check_error` call.

Thanks. Here's the result for you and @felipediel :

```
>>> devices[2].get_state()
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00;
\xcc\xfb\xff\xf4\xa4\xee\x03\xdb\x80\x0f\x87\x1b\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00'               
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/data/com.termux/files/home/python-broadlink/python-broadlink/broadlink/light.py", line 34, in get_state
    check_error(response[0x22:0x24])
  File "/data/data/com.termux/files/home/python-broadlink/python-broadlink/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.StorageError: [Errno -5] The device storage is full
```

felipediel, 2021-01-25:
Hi @kylemd. It's probably a similar firmware, that communicates almost the same, but without sending the payload length in the header. I am working on an update that will help us make it work. I will let you know when its ready. Thank you!

kylemd, 2021-01-25:
> Hi @kylemd. It's probably a similar firmware, that communicates almost the same, but without sending the payload length in the header. I am working on an update that will help us make it work. I will let you know when its ready. Thank you!

Thanks so much! If you need me to test anything let me know.

felipediel, 2021-10-17:
Thank you! :rocket:

closed 2021-10-17 by felipediel