mjg59/python-broadlink #525: Error when I try an install with python2.7

Bosseur31, 2021-01-26:
Hello, I can't install with python 2.7, yet in python 3 it works ... Here is the error

`pi@raspberrypi:~/python-broadlink $ sudo python setup.py install
running install
running bdist_egg
running egg_info
writing requirements to broadlink.egg-info/requires.txt
writing broadlink.egg-info/PKG-INFO
writing top-level names to broadlink.egg-info/top_level.txt
writing dependency_links to broadlink.egg-info/dependency_links.txt
reading manifest file 'broadlink.egg-info/SOURCES.txt'
writing manifest file 'broadlink.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/cover.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/device.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/exceptions.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/light.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/helpers.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/__init__.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/remote.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/switch.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/sensor.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/climate.py -> build/bdist.linux-armv7l/egg/broadlink
copying build/lib.linux-armv7l-2.7/broadlink/alarm.py -> build/bdist.linux-armv7l/egg/broadlink
byte-compiling build/bdist.linux-armv7l/egg/broadlink/cover.py to cover.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/cover.py", line 11
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/device.py to device.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/device.py", line 18
    timeout: int = 10,
           ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/exceptions.py to exceptions.pyc
byte-compiling build/bdist.linux-armv7l/egg/broadlink/light.py to light.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/light.py", line 24
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/helpers.py to helpers.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/helpers.py", line 5
    def calculate_crc16(input_data: bytes) -> int:
                                  ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/__init__.py to __init__.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/__init__.py", line 115
    dev_type: int,
            ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/remote.py to remote.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/remote.py", line 11
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/switch.py to switch.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/switch.py", line 12
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/sensor.py to sensor.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/sensor.py", line 17
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/climate.py to climate.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/climate.py", line 12
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/broadlink/alarm.py to alarm.pyc
  File "build/bdist.linux-armv7l/egg/broadlink/alarm.py", line 15
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

creating build/bdist.linux-armv7l/egg/EGG-INFO
copying broadlink.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying broadlink.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying broadlink.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying broadlink.egg-info/not-zip-safe -> build/bdist.linux-armv7l/egg/EGG-INFO
copying broadlink.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying broadlink.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
creating 'dist/broadlink-0.16.0-py2.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing broadlink-0.16.0-py2.7.egg
removing '/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg' (and everything under it)
creating /usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg
Extracting broadlink-0.16.0-py2.7.egg to /usr/local/lib/python2.7/dist-packages
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/__init__.py", line 115
    dev_type: int,
            ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/alarm.py", line 15
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/climate.py", line 12
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/cover.py", line 11
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/device.py", line 18
    timeout: int = 10,
           ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/helpers.py", line 5
    def calculate_crc16(input_data: bytes) -> int:
                                  ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/light.py", line 24
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/remote.py", line 11
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/sensor.py", line 17
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg/broadlink/switch.py", line 12
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

broadlink 0.16.0 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/broadlink-0.16.0-py2.7.egg
Processing dependencies for broadlink==0.16.0
Searching for cryptography==2.6.1
Best match: cryptography 2.6.1
Adding cryptography 2.6.1 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Finished processing dependencies for broadlink==0.16.0
`

Thanks you

KTibow, 2021-01-26:
Is there a reason you need Python 2? It's deprecated, so there's a low chance this issue will have action taken on it.

clientgithub, 2021-02-02:
> Is there a reason you need Python 2? It's deprecated, so there's a low chance this issue will have action taken on it.

There is one big reason - on old machines, python 2 runs 2-3 times faster. After the cryptography library was used, the script work much very slow.

KTibow, 2021-02-03:
@felipediel any opinions on this?

felipediel, 2021-02-04:
We no longer support Python 2.

KTibow, 2021-02-04:
So then is this issue a wontfix?

felipediel, 2021-02-04:
Sorry, this is a Python 3 module now. There is no way to "fix it" without rewriting the entire code.

closed 2021-02-04 by felipediel