mjg59/python-broadlink #531: RM4 Pro requires firmware update before accepting all WiFi passwords

deviantintegral, 2021-01-31:
I noticed that my RM4 Pro wouldn't connect to wifi with non-alphanumeric passwords before a firmware update. Is this common across many Broadlink products? If so, I'd be glad to update the README.

felipediel, 2021-03-14:
Thanks for reporting this. I think it would be nice to have a "TROUBLESHOOTING.md" where we put all the gotchas.

closed 2021-04-12 by felipediel

jspanitz, 2022-02-10:
> I noticed that my RM4 Pro wouldn't connect to wifi with non-alphanumeric passwords before a firmware update. Is this common across many Broadlink products? If so, I'd be glad to update the README.

I've had this problem since day one.  Are you saying there is a new firmware and it resolves the  WiFi password issues where it would not take special characters for the SSID pw?

If so, where did you get the firmware and how did you update the RM4 Pro?

PR #563 (2021-04-12): Add a TROUBLESHOOTING doc with WiFi password notes
Resolves #531 .