mjg59/python-broadlink #532: Control RM3 from webserver

bmscenter, 2021-02-03:
Hello,

Does anyone know how to achieve website control with our rm3 controller ??

I have apache, php and no have idea how to achieve it.

Please help

KTibow, 2021-02-03:
This is a Python library, not a PHP library. I would recommend installing Home Assistant instead.

bmscenter, 2021-02-04:
Thank you for your answer. I know this is a python library, but I would still like to script it over http anyway.
I already have a smart home at home and the only communication with it is through http requests.
I tried to execute the script using shell exec, but I don't know what the full syntax should look like then.

KTibow, 2021-02-04:
> I already have a smart home at home and the only communication with it is through http requests.

Could you elaborate? Personally I really like Home Assistant, and I would recommend you move to it.

bmscenter, 2021-02-04:
I have a Polish GRENTON system and an HTTP module.

KTibow, 2021-02-04:
> I tried to execute the script using shell exec, but I don't know what the full syntax should look like then.

What's the example that didn't work?

bmscenter, 2021-02-04:
exec('bash /home/pi/rm3/python-broadlink/cli/broadlink_cli --type 0x2737 --host 192.168.0.190 --mac 34ea34429fda --send @BED-TV.power');

I've got problem with good syntax

KTibow, 2021-02-04:
So does it work when you execute it in the command line? Specific error messages?

bmscenter, 2021-02-05:
Now i have code like this
<?php
    $command = escapeshellcmd('python3 /home/pi/rm3/python-broadlink/cli/broadlink_cli --type 0x2737 --host 192.168.0.190 --mac 34ea34429fda --send @BED-TV.power');
    $output = exec($command);
    echo $output;
?>

The command works because I added print test to the script.
However, unfortunately, it does not accept any parameters.

KTibow, 2021-02-05:
> However, unfortunately, it does not accept any parameters.

What do you mean?

bmscenter, 2021-02-05:
It runs the script but takes no arguments, nothing happens.
When I run a query on the command line, it works fine.

KTibow, 2021-02-06:
I don't know a lot about PHP; maybe if they have an official forum, you can ask for help there, and close this issue.

w35l3y, 2021-03-23:
Is the apache running at local network?

Consider reading parameters from `$_GET` or `$_POST`.

`<?php echo $_GET['param_name']; ?>`

That is how php handle them.
It doesn't have anything do with this library.

felipediel, 2021-10-18:
Possibly related to https://github.com/mjg59/python-broadlink/issues/194?