mjg59/python-broadlink #535: Error learn command (0x279d)

Bosseur31, 2021-02-09:
HI, I am using an API to communicate with this library, everything works fine but when I try to learn a command this error appears, does anyone have a idea ?

`File "/usr/local/lib/python3.7/dist-packages/gunicorn/workers/sync.py", line 175, in handle_request respiter = self.wsgi(environ, resp.start_response) File "/usr/local/lib/python3.7/dist-packages/falcon/api.py", line 269, in __call__ responder(req, resp, **params) File "/home/pi/broadlink-rm-rest/app/app.py", line 294, in on_put value = blaster.get_command() File "/home/pi/broadlink-rm-rest/app/db_helpers/blaster_db.py", line 95, in get_command value = device.check_data() File "/usr/local/lib/python3.7/dist-packages/broadlink/remote.py", line 26, in check_data return self._send(0x4) File "/usr/local/lib/python3.7/dist-packages/broadlink/remote.py", line 20, in _send check_error(resp[0x22:0x24]) File "/usr/local/lib/python3.7/dist-packages/broadlink/exceptions.py", line 137, in check_error raise exception(error_code) broadlink.exceptions.ReadError: [Errno -10] Read error`

Thanks you

KTibow, 2021-02-11:
Please use 3 backticks, not 1.

felipediel, 2021-02-15:
ReadError means you are trying to read an empty buffer. The CLI is a good starting point if you want to reimplement this: https://github.com/mjg59/python-broadlink/blob/b33bbdbbcf1270fb77f346f0e003273247d3ff2f/cli/broadlink_cli#L130-L144

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel