mjg59/python-broadlink #536: Learning 315 mhz codes

clientgithub, 2021-02-15:
Is it possible to learn not only 433 MHz, but also 315 MHz codes jon BroalLink RM2 Pro device? I have only 433 MHz device learning. Thank you.

KTibow, 2021-02-16:
This is probably a hardware issue.

clientgithub, 2021-02-16:
> This is probably a hardware issue.

on E-control software learning ok.

MM1969MM, 2021-11-06:
Hi There I use the RM4 Pro with app version 1.6.81 the app works fine and after providing the frequency band of 315 I got to get it capture my remote.
So far so good.

I also use the HOOBS homemaneger with the homebridge-broadlink-rm plugin. as I learned the plugin relies on the software of this project. When I use the "Scan Frequency" or "Learn" button of the plugin app, it does not recover the 315mhz frequency. I know this software here is not about the homebridge-broadlink-rm, but I hope that someone can explain to me if the 315mhz caputure is covered by python-broadlink?

Thanks and kind regards
Michael

RamEliC, 2021-11-11:
I recently started using the RM4 Pro and have the same issue.  The only work around (not a solution) that I could find to get it to work is to:

    1.  Use the Broadlink mobile app to start the process for learning a new function.  
    2.  In the last screen of the wizard before it starts scanning for the code, it'll ask what frequency you want to scan and you can input 315mhz.
    3.  Once you proceed, it will begin a 30 second countdown to scan for a signal.  Immediately initiate the RF learning using the CLI **after** the countdown begins.  It should then be able to pick up the signal.  

That being said, it would be nice to have an option to specify the frequency to scan within the CLI.  Another nice feature would be having an option to do a one step learn by recording the first signal it receives instead of having to do a frequency sweep first.

jrunning, 2022-07-09:
@RamEliC Thanks very much for your workaround. It’s the only way I was able to learn codes from my 315 MHz remote. I also wish python-broadlink let you specify a frequency.

closed 2024-04-21 by felipediel