mjg59/python-broadlink #541: RM4 Pro ( 0x653c ) partially discovered

Bosseur31, 2021-02-26:
Hello,

My device RM4 Pro ( 0x653c ) is partially discovered with `python3 broadlink_discovery`, the type, host, mac are discovered but the name is Unknown and the learning commands return an error. 

`pi@raspberrypi:~/python-broadlink/cli $ python3 broadlink_discovery
Discovering...
###########################################
Unknown
# broadlink_cli --type 0x653c --host 192.168.31.105 --mac a043b031959a
Device file data (to be used with --device @filename in broadlink_cli) : 
0x653c 192.168.31.105 a043b031959a

###########################################
RM2
# broadlink_cli --type 0x279d --host 192.168.31.83 --mac 780f77fd4e0b
Device file data (to be used with --device @filename in broadlink_cli) : 
0x279d 192.168.31.83 780f77fd4e0b
temperature = 0.0
`

`pi@raspberrypi:~/python-broadlink/cli $ python3 broadlink_cli --host 192.168.31.105 --mac a043b031959a --type 0x653c --learnfile test.txt
Traceback (most recent call last):
  File "broadlink_cli", line 131, in <module>
    dev.enter_learning()
AttributeError: 'device' object has no attribute 'enter_learning'
`

KTibow, 2021-02-26:
Check #480

Bosseur31, 2021-02-26:
Thanks you very much but Why this patch is not included in the master

KTibow, 2021-02-27:
It is.

closed 2021-02-27 by Bosseur31

ofirshtrull, 2021-02-27:
@KTibow as i saw it is in master but PyPi version wasn't changed
I suggest adding a giyhub action that automatically pushes to PyPi when the version has changed

KTibow, 2021-02-27:
So did I, but it needs an API key first.