mjg59/python-broadlink #543: Trying to discover device outside current subnet

Daan12345678, 2021-03-03:
Tried discovering a device outside the subnet my pc is using. Got an error.
```
❯ ./broadlink_discovery --ip 192.168.2.166                                                                                                            ─╯
Discovering...
Traceback (most recent call last):
  File "/home/daan/personal/broadlink/python-broadlink/cli/./broadlink_discovery", line 14, in <module>
    devices = broadlink.discover(timeout=args.timeout, local_ip_address=args.ip, discover_ip_address=args.dst_ip)
  File "/home/daan/.local/lib/python3.9/site-packages/broadlink/__init__.py", line 150, in discover
    return [gendevice(*resp) for resp in responses]
  File "/home/daan/.local/lib/python3.9/site-packages/broadlink/__init__.py", line 150, in <listcomp>
    return [gendevice(*resp) for resp in responses]
  File "/home/daan/.local/lib/python3.9/site-packages/broadlink/device.py", line 29, in scan
    conn.bind((local_ip_address, 0))
OSError: [Errno 99] Cannot assign requested address
```

```

Expected result

KTibow, 2021-03-03:
All it's doing is calling another library. Are you sure that's the right parameter?

Daan12345678, 2021-03-04:
You are correct, sorry for bothering you with this issue. 
Should have used this command:
```
./broadlink_discovery --dst-ip 192.168.2.166 
```

closed 2021-03-04 by Daan12345678