mjg59/python-broadlink #544: Authentication failed

supagu, 2021-03-03:
I checked out the git repo from here: https://github.com/mjg59/python-broadlink

then tried to run broadlink_discovery

but i get the following error:

[s@server cli]$ ./broadlink_discovery 
Discovering...
Traceback (most recent call last):
  File "/home/s/Projects/python-broadlink/cli/./broadlink_discovery", line 16, in <module>
    if device.auth():
  File "/usr/lib/python3.9/site-packages/broadlink/device.py", line 198, in auth
    check_error(response[0x22:0x24])
  File "/usr/lib/python3.9/site-packages/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthenticationError: [Errno -1] Authentication failed
[s@server cli]$ 

What does this mean and how to I get around the problem?

supagu, 2021-03-04:
I found this thread:
https://github.com/mjg59/python-broadlink/issues/501

I needed to go into the broadlink app and unlcok the device. This has fixed it.

closed 2021-03-04 by supagu

pinei, 2021-03-31:
> I found this thread:
> #501
> 
> I needed to go into the broadlink app and unlcok the device. This has fixed it.

How did you get this broadlink app?

KTibow, 2021-03-31:
From the Google Play store.