mjg59/python-broadlink #545: How to packet sniff to intercept commands?

supagu, 2021-03-04:
I have a BOSE amplifier, and my RM4 pro is able to turn it on and off via the broadlink app.
When I try to frequency scan using the cli it finds nothing.
Also the broadlink app cannot frequency scan the BOSE remote either!

So I want to work out the command being sent to my RM4 that makes it turn the BOSE on and off.

So far what I have tried is:

1. Block the RM4 from the internet
2. Installed wireshark on my mac
3. Install broadlink app on my android emulator which is running on my mac
4. Tried to use the broadlink app to send commands to my RM4, but it is not working.
Oddly, my phone says the device is offline but it still manages to send commands to make the RM4 do its thing.

Any ideas?
Having a wiki page on such things would be mighty helpful.

kylemd, 2021-03-04:
If you are using the Broadlink app it won't be able control the device if it's offline, as it requires to be able to talk to the Broadlink server to control them.

As for why the cli isn't working, follow the instructions on the readme, being:

1. Clone the repo to your computer (or Termux on Android if you're on your phone)
2. Run 'python3' from terminal
3. Follow the instructions in the [readme](https://github.com/mjg59/python-broadlink) and run those commands from the python terminal:

[image]

4. Run `print(devices)`. You should see your RM4. If you have more than one Broadlink device make a note of what number on the list your RM4 is, as you'll need to refer to it in the next step. The count starts as 0, e.g. if your RM4 was the second item on the list, then it would be #1.
5. If it shows **Locked=True** run `devices[x].set_lock(False)` to remove the lock. Using the example above, your RM4 would be `devices[1].set_lock(False)`
6. Run `devices[x].auth()` to authenticate with the device.
7. You can then run the rest of the commands from the [readme](https://github.com/mjg59/python-broadlink)

supagu, 2021-03-04:
Is there a way I can view what packets the RM4 will be getting in order to turn my BOSE on or off? 
The RM4 cannot learn the BOSE RF for some reason (not even the broadlink app detects the remote in learn mode) yet when in the broadlink app I select the BOSE (from the pre-known products) it is controlling my BOSE correctly. 
So I want to find what command that is being sent to my RM4 so I can send the command manually later on.
How do I do this?

KTibow, 2021-03-04:
It will also randomly reboot from my knowledge if it can't connect to the internet; I'm not sure if this was patched or not.

felipediel, 2021-03-12:
@supagu I extracted the codes for you.

## BOSE

### Type 1

```
# Power
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EjcSNxISEjcSEhISEjcSEhISEhISNxI3EhISEhI3EhISNxI3EhISEhI3EjcSEhI3EgAFAQ=='

# Mute
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EjcSNxISEjcSEhISEjcSEhI3EhISEhISEhISEhISEhISEhI3EjcSNxI3EjcSNxI3EgAFAQ=='

# Arrow up
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EjcSNxISEjcSEhISEjcSEhI3EhISEhI3EjcSEhISEhISEhI3EjcSEhISEjcSNxI3EgAFAQ=='

# Arrow left
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EjcSNxISEjcSEhISEjcSEhISEhISEhISEhISNxISEjcSNxI3EjcSNxI3EhISNxISEgAFAQ=='

# Arrow right
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EjcSNxISEjcSEhISEjcSEhISEhISEhISEhISNxI3EhISNxI3EjcSNxI3EhISEhI3EgAFAQ=='

# Arrow down
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EjcSNxISEjcSEhISEjcSEhISEhISEhI3EjcSEhISEhISNxI3EjcSEhISEjcSNxI3EgAFAQ=='

# Volume down
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EjcSNxISEjcSEhISEjcSEhISEjcSEhISEhISEhISEhISNxISEjcSNxI3EjcSNxI3EgAFAQ==',

# Volume up
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EjcSNxISEjcSEhISEjcSEhI3EjcSEhISEhISEhISEhISEhISEjcSNxI3EjcSNxI3EgAFAQ=='
```

### Type 2
```
# Power
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EhISEhISEhISEhI3EhISNxISEhISNxI3EhISEhI3EhISNxI3EhISEhI3EjcSEhI3EgAFSg=='

# Mute
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EhISEhISEhISEhI3EhISNxISEhISEhISEjcSEhISEjcSNxI3EjcSNxISEjcSNxISEgAFSg=='

# Arrow up
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EhISEhISEhISEhI3EhISNxISEhISEhISEhISNxISEjcSNxI3EjcSNxI3EhISNxISEgAFSg=='

# Arrow left
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EhISEhISEhISEhI3EhISNxISEhISEhISEjcSEhISEjcSNxI3EjcSNxISEjcSNxISEgAFSg=='

# Arrow right
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EhISEhISEhISEhI3EhISNxISEhISEhISEjcSEhISEjcSNxI3EjcSNxISEjcSNxISEgAFSg=='

# Arrow down
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EhISEhISEhISEhI3EhISNxISEhISEhISEhISEhI3EjcSNxI3EjcSNxI3EjcSEhISEgAFSg=='

# Volume down
'JgBIAAABJ5QSEhI3EhISNxI3EjcSEhI3EhISEhISEhISEhI3EhISNxI3EjcSEhISEhISEhISEhISEhISEjcSNxI3EjcSNxI3EgAFSg=='
```

### Type 3
```
# Power
'JgBQAAABJ5QUEBM2FQ4VNBU1FDYUEBM2FDYVNBUOFTQVDhUPEzYUEBMQExATNhU0FQ4VDhU1FBATNhQ2FBAVDhU0FTUVDhU1FAAFMAABI0gUAAzNAAAAAAAAAAA='

# Mute
'JgBQAAABJ5QVDhU0FQ4VNRQ2FDYUEBU0FjQVNBUPEzYUEBMQEzYVDhU0FQ8TDxUOFQ8TEBMQExATEBM2FTQVNBU0FTUUNhQ2FAAFMAABI0kUAAzNAAAAAAAAAAA='

# Arrow up
'JgBQAAABJ5QWDhU0FQ4VNRQ2FDYUEBU0FTQVNBUPEzYUEBMQEzYVDhUOFQ4VDhUOFQ4VDxM2FDYUNhU0FTQWNBU1FDYUEBMQEwAFMAABI0kTAAzNAAAAAAAAAAA='

# Arrow left
'JgBQAAABJ5QVDxQ1FQ4VNRQ2FDYUEBU0FjQVNBUPEzYUEBMQEzYWDhUOFQ4VDhUOFQ4VNRQQEzYUNhU0FTQVNBY1FBATNhQQEwAFMAABI0gUAAzKAAAAAAAAAAA='

# Arrow right
'JgBQAAABJ5QVDhU0FQ4VNRQ2FDYUEBU0FTQVNBYPEzYUEBMQEzYVDhUOFQ4VDhUOFQ4VNRQ2ExATNhY0FTQVNBU1FBATEBM2FAAFMAABI0gUAAzMAAAAAAAAAAA='

# Arrow down
'JgBQAAABJpUVDxQ1FQ8UNhQ2EzYUEBU0FTQVNBUPEzYUEBMQEzYVDhUOFQ4VDhUOFQ4VNRQQExATNhQ2FjQVNBU1FBATNhQ2FAAFMAABI0gUAAzNAAAAAAAAAAA='

# OK
'JgBQAAABJZcUEBM2FBATNhQ2FDYUERM2FDYUNhQREzYUERMREzYUEBMQExETERMQExATNhQ2FDYUNhQ2FDYUNhQ2FBETERMREwAFMgABIkkUAAzMAAAAAAAAAAA='

# Volume down
'JgBQAAABJ5QVDhU0FQ8VNRQ2FDYUEBU0FTUVNBUPEzYUEBMQEzYWDhUOFTQVDhUOFQ8TEBMQExATNhQQFDUVNBU0FTUUNhQ2FAAFMAABI0gUAAzKAAAAAAAAAAA='

# Volume up
'JgBQAAABJ5QVDhQ1Fg4VNRQ2FDYUEBQ1FjQVNBUPEzYUEBMQEzYVDhQ1FTUUDxUPExATEBMQExATEBMQFTQVNBU0FTYUNhQ2FAAFMAABI0gUAAzLAAAAAAAAAAA='

# Menu
'JgBQAAABJ5QVDxQ1FQ4VNRQ2FDYUEBU0FTQVNBUPEzYUEBMQEzYVDhUOFQ4VDhUOFTUUEBMQEzYUNhU0FTQVNBUPEzYUNhQQEwAFMAABI0gUAAzNAAAAAAAAAAA='
```
Please don't tell my mom I'm doing this. She wanted me to be a doc.

closed 2021-03-12 by felipediel

supagu, 2021-03-15:
Thanks! that is working for me using the broadlink_cli (once I convert it from base64 to hex).
It would be really helpful to have as part of the wiki how to get such data