mjg59/python-broadlink #548: RM Mini 3 - auth_check_error

skinnerstein, 2021-03-08:
I've got my RM Mini 3 on my network. When I ran discover the device was found and added into the devices[] array. 
[<rm4: Broadlink RM mini 3 (0x5f36) at 192.168.86.54:80 | 24:df:a7:36:84:39 | 智能遥控 | Locked>]

Now, when I run devices[0].auth, I get the following:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python39\lib\site-packages\broadlink\device.py", line 199, in auth
    check_error(response[0x22:0x24])
  File "C:\Python39\lib\site-packages\broadlink\exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthenticationError: [Errno -1] Authentication failed

KTibow, 2021-03-09:
You need to unlock it first (either use the Broadlink app settings, or factory reset and add it again without locking). If that fixes, consider closing.

skinnerstein, 2021-03-09:
That fixed the issue, thanks!