mjg59/python-broadlink #556: SCB2 wifi switch

MichielioZ, 2021-03-15:
About a year ago got this wifi switch imported from America as Europe didn't have this "NoApp" version at the time...
As far as I can see it's just the MCB1 repackaged with "NoApp" support ?!
Maybe it already works ? Just thought I had to mention it is out there...
I didn't test anything as it is currently used by my parents with Google Home.
I have an spare one laying around here in case anyone needs testing...

[image]
[image]

KTibow, 2021-03-15:
NoApp just means that it sets up a network where you can enter your network credentials without needing an app. It could be that it _did_ have it, but they didn't call it NoApp then.

MichielioZ, 2021-03-15:
@KTibow I specifically imported them because of the "NoApp" to make it easier for my parents to install it. So yeah, I know what it's for... They just seem 100% the same (visually at least) to the MCB1, but if they identify under the new name SCB2 then this Python script won't recognize them unless you add the name and point to the MCB1 device for functionality. Currently I'm not using it with HA, but maybe somebody else (or me in the future) could use the new name being recognized...

felipediel, 2021-03-15:
Hi @MichielioZ. Could you please discover this device so we can add it?
```python3
import broadlink as blk
d = blk.discover()
print(d)
```

MichielioZ, 2021-03-15:
Not much information from discover:
```
[<device None None (0x6494) at 192.168.178.123:80 | c8:f7:42:75:b0:47 | 25748 | Locked>]
```
I unlocked it in the Broadlink app, unlocked it through d[0].set_lock(False), but it keeps showing up as locked...
Node-RED discover gives me:
```
typeid: "6494"
type: "Broadlink: Unconfigured Device Type: Log an issue with the device details and typeid returned as this may be a new device type."
name: "25748E��E�c[;R�C��"
```

On a side note: apparently my Smart Bulb (LB27 R1) which previously showed up in discover now doesn't anymore... weird

felipediel, 2021-03-15:
I think we can control this device with the sp4b class. Let's do a test:
```python3
import broadlink as blk

d = blk.sp4b(('192.168.178.123', 80), 'c8f74275b047', 0x6494)
d.auth()
print(d.get_state())
d.set_power(True)
d.set_nightlight(True)
print(d.get_state())
```

MichielioZ, 2021-03-15:
get_state: `{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'childlock': 0}`
I can now turn the switch on and off with set_power.
set_nightlight obviously does nothing here (it's physically an MCB1 wifi switch)...

felipediel, 2021-03-15:
Thank you! It works. I think it is reasonable to create a separate class to improve duck typing. I will do it.

MichielioZ, 2021-03-15:
@felipediel No, thank you ! This was resolved really fast :-)
I now edited my __init__.py so until next update of HA I can have a working switch (that I don't actually need, haha).
I am sure other people will thank you in the future for adding this and eventually I'm gonna use the switch...

felipediel, 2021-10-18:
Thank you! :rocket:

closed 2021-10-18 by felipediel