mjg59/python-broadlink #559: Is there any easier way to "backup" codes already learned in the app from certain device at once ?

w35l3y, 2021-03-22:
Currently, I am backing up each code, one by one, with 2 RM4C MINI side by side, one as a sender and another as receiver.

The sender have all codes in the app.
And the receiver, I am using the command `broadlink_cli --device "0x62be 1.1.1.1 aaaaaaaaaaaa" --learnfile FILENAME`.

felipediel, 2021-03-22:
This script works with e-Control: https://github.com/NightRang3r/Broadlink-e-control-db-dump. I don't know if it works with the new Broadlink app.

I think the most user friendly solution would be to download the codes directly from the cloud. It is possible, but it needs investigative work that I will never find time to do. I leave it as a suggestion for anyone who wants to contribute to the project, it would be a great addition.

felipediel, 2021-03-22:
Might be useful: https://docs.ibroadlink.com/public/configuration-sdk+ctc/rm_related_api/.

w35l3y, 2021-03-23:
> Might be useful: https://docs.ibroadlink.com/public/configuration-sdk+ctc/rm_related_api/.

Very promising.
I don't understand if I made something wrong, but the domain in the docs didn't work. I will contact them.

https://xxxbizopenplatform.ibroadlink.com/openproxy/v2/ircode/info

Thanks!
Currently, I have successfully copied all codes I needed.

spartakos87, 2021-07-12:
> > Might be useful: https://docs.ibroadlink.com/public/configuration-sdk+ctc/rm_related_api/.
> 
> Very promising.
> I don't understand if I made something wrong, but the domain in the docs didn't work. I will contact them.
> 
> https://xxxbizopenplatform.ibroadlink.com/openproxy/v2/ircode/info
> 
> Thanks!
> Currently, I have successfully copied all codes I needed, one by one.

Hello! Did you find the valid url?

w35l3y, 2021-07-12:
> Hello! Did you find the valid url?

I gave it up. I contacted them and they require a developer license. It is USD 2000 per account.

felipediel, 2021-10-18:
USD 2000 per account! Closing :laughing:

closed 2021-10-18 by felipediel