mjg59/python-broadlink #562: BroadLink BestCon MSK1 integration

redotheone, 2021-03-26:
Do you think it is possible to add MSK1 to python-broadlink? 

Also, how do we know what is the hex id for a given device? 

[image]

KTibow, 2021-03-26:
> Do you think it is possible to add MSK1 to python-broadlink?

Probably. Have you heard of that before @felipediel?

> Also, how do we know what is the hex id for a given device?

If you're asking for how to find it, once connected to your network, try this:
```
import broadlink
devices = broadlink.discover() # This will take 10 seconds
print(devices)
```
The `0xblabla` is the hex id for each device.

felipediel, 2021-03-26:
Probably the same as S1C, but with special payload headers.

CyberMod-Studio, 2022-03-31:
Can MSK1 works with Home Assistant?

kevinelliott, 2022-05-04:
```
>>> broadlink.hello('192.168.14.200')
broadlink.device.Device(('192.168.14.200', 80), mac=b'$\xdf\xa7\xacU\x90', devtype=11001, timeout=10, name='IHG', model='', manufacturer='', is_locked=False)
```

kevinelliott, 2022-05-04:
So there's the hex id for the Broadlink MSK1 BestCon hub. `0x2af9`

kevinelliott, 2022-05-04:
@KTibow @felipediel @redotheone

felipediel, 2024-04-21:
Thanis @kevinelliott! Does it work with the SC1 class? If so, we can add the new product id.