mjg59/python-broadlink #571: Broadlink LB1 0x60c8 unable to authenticate

goncalovalverde, 2021-04-03:
Hi.

I've been trying to use this module with a Broadlink LB1 0x60c8 bulb but I'm getting an error while trying to authenticate.
This is the response that I'm getting when I do a device.auth:
'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00;\xcc\xff\xff\xc8`\xe9\x03 \xa1\xd0;\xb2\xb0C\xa0\x00\x00\x00\x00\xaf\xbe\x00\x00'

This fails on the check_response

Any pointers?

KTibow, 2021-04-03:
Could you please format your data using backticks instead of quotes? @felipediel do you know anything about this?

felipediel, 2021-04-03:
The device is locked. You need to unlock the device in the official app.

goncalovalverde, 2021-04-03:
I had the bulb configured using the magic home - smart home app which is the one recommended on the box of the bulb.
The info from doing a discover stated that the bulb was unlocked
Anyway I downloaded the broadlink app and after reseting the bulb I managed to send the wifi configuration. 
The broadlink app complained about the region and the lamp doesn't appear in the application. 
Nevertheless this seems to have done the trick to unlock the lamp and now I can authenticate and control the lamp via python.
Thank you for your feedback

felipediel, 2021-04-03:
You're welcome!

closed 2021-04-03 by felipediel

felipediel, 2021-04-05:
@goncalovalverde I know the issue is resolved, but do you mind helping me understand why the lock status is incorrect?

```python3
import socket

with socket.socket(socket.AF_INET, socket.SOCK_DGRAM) as conn:
	packet = bytearray(0x30)
	packet[0x26] = 6
	conn.sendto(packet, ('192.168.0.16', 80))  # Example LB1 IP address
	conn.settimeout(5)
	print(conn.recvfrom(2048))
```
What is the output?

goncalovalverde, 2021-04-06:
Sure.
Here goes the output:

(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00>\xcc\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc7`\x94\x01\xa8\xc0\xc3jL\xb0C\xa0Smart Bulb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x01E\x9a\x9aE\x02\x00\x00\x00xs\x1dF\x8f\x0c\x00\x00\x00\x00', ('192.168.1.148', 80))