mjg59/python-broadlink #578: SCB1E missing energy

shyrwall, 2021-04-13:
Hi

SCB1E is mapped to sp4b but sp4b does not contain an energy class. 

Copying get_energy function from sp3s results in : 

broadlink.exceptions.StorageError: [Errno -5] The device storage is full

shyrwall, 2021-04-13:
Hmm. Maybe get_energy was deprecated? :) get_state worked.

KTibow, 2021-04-13:
This is an error talked about many times before. What is the get_energy source?

shyrwall, 2021-04-16:
Sorry for late reply. Yes I saw it was talked about before but opened the issue because it looked like it didn't have a way to get the watt usage etc but it seems it was just baked into get_state() so maybe get_energy() was deprecated for this device. 

Anyway. Everything working so closing the issue.

closed 2021-04-16 by shyrwall