mjg59/python-broadlink #580: Unsupported device: 0x4e2a Error

Stefos13, 2021-04-16:
Hello, I have an RM Mini and I'm getting the following error. The commands are being sent without any problems and the device works. Everything is updated to the latest version.

> Logger: homeassistant.components.broadlink.config_flow
Source: components/broadlink/config_flow.py:41
Integration: Broadlink (documentation, issues)
First occurred: 12:51:23 PM (1 occurrences)
Last logged: 12:51:23 PM
Unsupported device: 0x4e2a. If it worked before, please open an issue at https://github.com/home-assistant/core/issues

KTibow, 2021-04-16:
Which type of RM Mini?

Stefos13, 2021-04-16:
It's Rm Mini 3.. I don;t know if you ask for something else. If yes where can i find the type of the rm mini?

KTibow, 2021-04-18:
Reboot HA, then see if it's still there.

Stefos13, 2021-04-18:
I have done it many times.. Still there

felipediel, 2021-04-19:
This error message will disappear after https://github.com/home-assistant/core/pull/49394.

We don't have 0x4e2a in our code base. What device is this?

Stefos13, 2021-04-19:
> This error message will disappear after [home-assistant/core#49394](https://github.com/home-assistant/core/pull/49394).
> 
> We don't have 0x4e2a in our code base. What device is this?

The device is the Broadlink RM Mini 3. Maybe i can check the device itself later to see if i can find anything extra

felipediel, 2021-04-19:
Interesting, I had never seen an RM mini 3 in this range. Forget what I said above. Could you please confirm that it works with the rmminib class?

```python3
import broadlink as blk
d = blk.rmminib(('192.168.0.16', 80), 'aabbccddeeff', 0x4e2a)
d.auth()
d.send_data(bytes(range(255)))
```

Does the LED blink?

Stefos13, 2021-04-19:
> Interesting, I had never seen an RM mini 3 in this range. Forget what I said above. Could you please confirm that it works with the rmminib class?
> 
> ```python
> import broadlink as blk
> d = blk.rmminib(('192.168.0.16', 80), 'aabbccddeeff', 0x4e2a)  # Example IP and MAC address
> d.auth()
> d.send_data(bytes(range(255)))
> ```
> 
> Does the LED blink?

I will try it once i get home but I need you to tell me how to try this. Just run it as python script from any device in my network I guess, and the IP of the device.

felipediel, 2021-04-19:
Yes.

Stefos13, 2021-04-19:
I tried it with Python 3.7 and i get this error 
"Traceback (most recent call last):
  File "t.py", line 2, in <module>
    d = blk.rmminib(('192.168.1.4', 80), 'aabbccddeeff', 0x4e2a)  # Example IP and MAC address
AttributeError: module 'broadlink' has no attribute 'rmminib'"

Don;t know if I am missing something or the problem is somewhere else..

felipediel, 2021-04-19:
Your library is not up to date. Try using the rm4 class.

Stefos13, 2021-04-20:
Haven't managed to make it work yet.. I have installed broadlink etc and I'm getting "ImportError No module named broadlink"
I will try to setup and VM with linux and do it from there

Stefos13, 2021-04-20:
I installed ubuntu , installed broadlink and im getting this error now
[image]

felipediel, 2021-04-20:
Ok, this is not an rm4. Let's try the rm class:
```python3
import broadlink as blk
d = blk.rm(('192.168.0.16', 80), 'aabbccddeeff', 0x4e2a)  # Example IP and MAC address
d.auth()
d.send_data(bytes(range(255)))
```

Stefos13, 2021-04-20:
Ok that worked, the light blinked!

felipediel, 2021-04-20:
Awesome. I am creating a PR right now.

felipediel, 2021-04-20:
Done. If you want to patch your installation before the release:

1. Open Home Assistant and install "SSH & Web Terminal" add-on
2. Disable Protection mode
3. Click Terminal on the sidebar and install the patch:
```
docker exec -it homeassistant /bin/bash
pip3 install git+https://github.com/felipediel/python-broadlink.git@0x4e2a --force-reinstall --no-deps
```
4. Restart the server.

You need to reinstall the patch every time you update Home Assistant.

felipediel, 2021-04-20:
Just to confirm... Is [this](https://www.amazon.com.br/BroadLink-RM-Mini-Compat%C3%ADvel-Assistente/dp/B01FK2SDOC/ref=asc_df_B01FK2SDOC/?tag=googleshopp00-20&linkCode=df0&hvadid=379728501964&hvpos=&hvnetw=g&hvrand=2494668768212166995&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1001686&hvtargid=pla-318070470702&psc=1) your device?

Stefos13, 2021-04-21:
> Just to confirm... Is [this](https://www.amazon.com.br/BroadLink-RM-Mini-Compat%C3%ADvel-Assistente/dp/B01FK2SDOC/ref=asc_df_B01FK2SDOC/?tag=googleshopp00-20&linkCode=df0&hvadid=379728501964&hvpos=&hvnetw=g&hvrand=2494668768212166995&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1001686&hvtargid=pla-318070470702&psc=1) your device? I checked all the apps (Broadlink, IHC and e-Control) and the device is not there. Are you sure about the type?
> 
> ```python
> import broadlink as blk
> d = blk.hello("192.168.0.16")  # Example IP address
> print(d)
> ```
> 
> What is the output?

Yeap, that's the device. I will try it later because i am at work now and i will let you know

Stefos13, 2021-04-21:
> Just to confirm... Is [...]

Here you are
智能遥控 (Broadlink RM mini 3 0x2737 / 192.168.)

felipediel, 2021-04-21:
Your RM mini 3 is a 0x2737. As I suspected, 0x4e2a is a different device with Broadlink DNA. Maybe an AC unit?

danielchalmers, 2022-07-26:
> Closing, if you figure out what device is 0x4e2a, please let us know! Thank you!

@felipediel you're right, 0x4e2a is indeed an AC unit. I found the same code for mine

lastsoft81, 2023-09-23:
hello.
so I came across this thread as im looking to connect my AUX AC UNIT to my homeassitant with no luck.
it has the same id 0x4e2a and HA says not supported when im trying to add it.
is is possible to add this ac unit to HAOS?

Best regards,
Shlomi Chen.