mjg59/python-broadlink #596: RM4C pro not supported

kst84, 2021-06-04:
Can we please have a RM4C pro added ? 
[homeassistant.components.broadlink.config_flow] Unsupported device: 0x6184

im0t0, 2021-07-08:
Home Assistant core-2021.7.0

2021-07-08 11:54:00 ERROR (MainThread) [homeassistant.components.broadlink.config_flow] Unsupported device: 0x6184. If it worked before, please open an issue at https://github.com/home-assistant/core/issues

im0t0, 2021-08-04:
core-2021.8.0
Still not added.

DutchessNicole, 2021-08-23:
I would also love to have this added to the library. Do you need any info from us?

Pasha1402, 2021-08-24:
+1. Good alternative to more expensive and huge RM4 Pro.

DutchessNicole, 2021-09-08:
in case of HassOS that would be /usr/local/lib/python3.9/site-packages/broadlink/__init__.py

felipediel, 2021-10-17:
Thank you! :rocket:

closed 2021-10-17 by felipediel

mikkelsiggaard, 2023-03-06:
@felipediel Why was this closed? 
It still needs work in code to be supported?

felipediel, 2023-03-06:
Hi. It's already supported.

mikkelsiggaard, 2023-03-07:
So this step is no longer necessary? 

> For now HA users can modify __init__.py in HA root PY folder /lib/python3.8/site-packages/broadlink/__init__.py by adding under SUPPORTED_TYPES section the following line 0x6184: (rm4pro, "RM4C pro", "Broadlink"),

felipediel, 2023-03-07:
Nope