mjg59/python-broadlink #601: please ! add support for 0x6184

givanildo, 2021-07-09:
0x6184: (rm4pro, "RM4 pro", "Broadlink"),

im0t0, 2021-08-04:
core-2021.8.0
Still not added.

MichielioZ, 2021-08-20:
> 
> 
> core-2021.8.0
> Still not added.

Yeah, I don't think anybody has been working on this script in a few months now...
Temporary solution is to manually add it to the list, but it will have to be re-added every time after reboot 😞

givanildo, 2021-08-20:
No problem ! I'm used to it!

im0t0, 2021-09-05:
core-2021.9.3
Still not added.

Project is dead?

MichielioZ, 2021-09-14:
> 
> 
> core-2021.9.3
> Still not added.
> 
> Project is dead?

I don't think the project is dead but I fear it may be shifted more towards "Home Assistant Only" territory...
The most active developer a couple of months ago was @felipediel and he was really active at the time.
At the time he was starting to implement a new version (in dev) which would be a better implementation of the Broadlink protocol, but the work iover there has been halted too.
This repository is not directly related to Home Assistant (HA) but is a more a standalone python solution that happens to get used by HA.
I think people at HA wanted @felipediel to put his efforts more towards implementing his ideas directly into HA core instead of improving this repository...
Then again, he could also just be busy with other things...

I wish I had the Python knowledge (and time) to work on this repository myself.
Maybe some time in the future, but that won't help you now ;-)

givanildo, 2021-09-14:
for the inclusion of this device just add this line:
0x6184: (rm4pro, "RM4 pro", "Broadlink"),
in the file __init__.py . 
Inside /./usr/local/python3.9/BroadLink/. I don't see any work to do this . But, alright, let's keep adding manually .

im0t0, 2021-09-14:
> for the inclusion of this device just add this line:
> 0x6184: (rm4pro, "RM4 pro", "Broadlink"),
> in the file **init**.py .
> Inside /./usr/local/python3.9/BroadLink/. I don't see any work to do this . But, alright, let's keep adding manually .

HA is frequently updated, tired of constantly adding manually

felipediel, 2021-09-18:
Hi folks. The project is not abandoned. If you have a look into the dev branch of this repository or Home Assistant's Broadlink integration, you'll see that it continues to be maintained regularly, at a slower pace (much slower than before, I agree).

The device 0x6184 has already been added to the dev branch. The release is taking a while because I've been busy, I have tight deadlines to meet, I can't change my focus right now without losing my job.

I'm negotiating for a week to participate in the Hacktoberfest, in which I plan to finish v0.18 and take it to Home Assistant. Until then, people need to be patient and understand that this is an open-source project under the MIT license. Updates do take time, not out of spite, but because we have to get the food on the table first.

givanildo, 2021-09-18:
Tks a lot !!! 

Att.: Givanildo Brunetta
            66999552704

felipediel, 2021-10-17:
Thank you! :rocket: