mjg59/python-broadlink #606: sending repeat keys

spacelama, 2021-07-12:
If I invoke broadlink_cli repeatedly, there is quite a slow upper limit to the number of commands, such as volume down, that I can send per second.  On the other hand, I can set up broadlink_cli to capture an IR remote control while holding down the button, and it only ceases to capture when I release the button.  I can then send that learnt command, and it replays the button being held.  But it replays it exactly of course - with all the timing quirks of the original sent command (I just told broadlink_cli to learn me pressing volume down and volume up long enough to get 10 "clicks" of volume.  I can replay them 5 times and get exactly 50 clicks of volume, but I can see the cyclical nature of how it sped up and slowed down randomly while being pressed during that learning cycle).

It looks like from the RM4 protocol set, there's a repeat value.  Does it work in that the RM4 sends that command in a repeating fashion without repeated setup/teardown of the infrared signal, and how do we teach broadlink_cli to make use of that repeat value so we don't in turn have the costs of the repeated setup/teardown of python?

KTibow, 2021-10-03:
Couldn't you make your own Python script without using the CLI to make it faster?

spacelama, 2021-10-04:
I think the real overhead is in the header of the IR commands - you can see it in the LED flashing pattern.  10 short presses takes a lot longer to issue than 1 long press equalling 10 repeating button presses.