mjg59/python-broadlink #610: [Help requested] every time I press the same button on the remote, the learned code changes

Zen3515, 2021-08-12:
I use the following code to learn my AC remote. The problem here is that the codes it returns keep changing, for example, I try to learn the off code, this is the last 10 output.

```python
def get_packet(device: broadlink.remote.rmmini, TIMEOUT=30):

    device.auth()
    device.enter_learning()

    packet = None

    # Fix according to https://github.com/mjg59/python-broadlink/blob/29345a129ff6c98962fccba8ada1ea049fd9ec16/cli/broadlink_cli#L222-L232
    start = time.time()
    waiting_times = 0
    while time.time() - start < TIMEOUT:
        time.sleep(1)
        try:
            packet = device.check_data()
            print("PRESSED!!")
        except (ReadError, StorageError):
            continue
        else:
            break
        finally:
            waiting_times = (waiting_times + 1) % 4
            print('|/-\\'[waiting_times] + ' waiting', end='\r')
    else:
        print("No data received...")
        breakpoint()
        exit(1)

    if packet:
        return b64encode(packet).decode('utf8')
```

The same button is being pressed 10 times. Each time it returns a new code.
```
JgAqAZCOFDMSNRI1EzQSEhIREzQSEhMQExETEBISEzQSNRIREzQTERMQExETEBMRExATNBI1EjUSNRM0EjUSNRI1EhITEBMRExATERIREhITEBMREzQSNRIREzQUMxMRExATERMQExETEBMRExATERM0ETYTNBMQExETEBISExATERMQEhITNBIREzQTNBISEzQTNBIRE9uSjhI1EjUSNRI1EhITEBM0ExESERISExATERI1EzQTEBM0FBATEBMRExASEhIREzQSNRI1FDMSNRI1EjUSNRQQExATERMQEhITEBISExATERMxFDYSERM0FDIVEBMQExETEBMRExATERIRExETMxM1ETYTEBMREhETERMQExESERMREzQTEBI1EjUTERM0EzMTERMADQU=
JgAqAY6QFTIVMhI1EjUVDxQPFTIVDxQPFQ8UDxUPFDMSNRMQFTIVDxQPFQ8UDxUPFA8VMhI1EjUVMhUyEjUVMhQzFBAUDxUPFA8VDxQPFQ8UDxUPFDMUMxQPFTISNRQQExAUEBQPFQ8UDxUPFA8VDxQzFDMUMxQPFQ8UDxQQFA8VDxQPFQ8UMxQPFDMSNBYPFDMUMxIRFNqQkBQzFDMSNRI1FBATEBQzEhIUDxQQExAUEBQzEjUSERUyFBAUDxQQFA8UEBQPFDMUMxQzFDMSNRI1FDMSNRQQFA8VDxQPFBAUDxQQFA8UEBQzFDMSERQzEjUUEBQPFBAUDxQQFA8UEBQPFBAUMxQzFDMUDxQQFA8UEBQPFBAUDxQQFDMSERQzEjUUEBM0EjUREhQADQU=
JgAqAZCOFDMSNRUyFTISEhQQFDIVDxQPFBAUDxUPFDMSNRQQFDIVDxQPFQ8UDxQQExEUMhI1FDMTNBI1EzQTNBQzExEUDxQQExEUDxMRExAUEBMQFDMTNBIRFTQRNBUPExATERMQFBATEBMRExETEBM0EzQUMxMQExETEBMRExAUEBMQExEUMxMQEzQUMxISEzQUMxQPE9uRjxM0EzQUMxM0ExETEBQzExETEBQQExAUEBM0EzQSERM0FBATEBMRExATERMQEzQTNBM0FDMTNBI1FTITNBMRExATERMQExETEBMRExATERM0EzQTEBM0FDMTERQPExETEBMRFA8TERQPExETNBM0EzQTEBMRExATERMQExETEBMRFDMUDxQzFDMTERM0EzQSERMADQU=
JgAqAZGNFTMUMxQzFDMUDxUPFDMUDxUPFA8VDxQPFDMSNRISFDMSERUPFA8VDxQPFBAUMxQzEjUSNRI1FDMSNRQzFA8VDxQPFQ8UDxQQFA8VDxQPFTIVMhITEzMSNRQPFQ8UDxQQFA8VDxQPFBAUDxUyFDMVMhUPFA8UEBQPFBAUDxUPFA8VMRMSFDMSNRQPFDMUMxISFNmTjxMzFDMRNhI1ERIUEBQzFA8VDxQPFBAUDxQzFTIUEBQzFA8UEBQPFBAUDxQQFDMUMxI1FDMUMxQzETYUMxESFBAUDxQQFA8UEBQPFBAUDxQzEjUUEBQzFDMUDxQQFA8UEBQPFBAUDxQQFA8UMxI1FTIUEBQPFBAUDxQQFA8UEBQPFTISEhQyEjYREhQzFDMUEBQADQU=
JgAqAZGMEzUSNRQzEjUSEhQPFTIVDxQPFQ8UDxUPFDMSNRIRFTISEhQPFQ8UDxQQFA8VMhQzEjUSNRI1EjQTNRUyFBAUDxUPFA8VDxQPFQ8UDxUPFDMSNRQPFTISNRUPFA8VDxQPFQ8UDxQQFA8VDxQzEjEWNRQPFQ8UDxQQFA8VDxQPFQ8UMxQPFTISNRQQFDMSNRIRFdmTjRUyEjUVMhI1FBAUDxQzFBAUDxQQFA8UEBQzFDMSERQzEhIUDxQQFA8UEBQPFDMSNRQzFTISNRQzEjUVMhQQFA8UEBQPFBAUDxQQFA8UEBQzETYUDxQzFDMVDxQPFBAUDxQQFA8UEBQPFQ8UMxE2EjUSERQQFA8UEBQPFQ8UDxQQFDMUDxQzEjUUEBQzFDISEhQADQU=
JgAoARM0EzQTNBM0ExATERM0ExATERMQExETEBM0EjUTERM0ERITERMQExETEBMREzQTNBM0EjUSNRQzEjUTNBMQExETEBMRExATERMQExETEBM0EzQTERMyEzYUDxISExATERMQExETEBMRExATNBM0EjUTERMQExETEBMRExATERMQEzQTERM0EjUSERM0EzQTERPako8SNRI0FTIVMhMRExETMxMRExETEBMRExATNBI1ExETNBMQExETEBMRExATERMzEzQUNBI1EjUUMxM0EjQSEhMRExATERMQExETEBMRExATNBI1EhITMxQzEhITERIRExETEBMRExATERMQEzQTNBM0EhITEBMREhETERMQEhITEBM0ExAUMxQzFBATNBIzFRETAA0F
JgAqAY6PEzUTNBI1FDMTERMQEzQSEhIRExESERMREzQTNBESEzQUEBMQExETEBMRExATNBI1EjUTMxUzEjUUMxQzEhITEBMRExATERMQFBASERMREzQTNBMQEzQTNBMRExATERQPExETEBMRExATERM0EzQRNhQPExETEBMRExATERMQExETNBMQEzQUMxQQEzQRNhQPE9qRkBM0EjUSNRI1EhITEBM0EhITEBMRExATERM0ETUSEhM0FBATEBMRExATERMQFDMSNRQzEzEXMhM1EjUUMxIRFBATERMQExETEBMRExATERI1EzQTEBMzFDQSEhMQExETEBMRExATERMQExETMxI1EjYTEBMRExATERMQExETEBMREzMTERQzEjUUEBMzFDMSEhIADQU=
JgAqAZCOFTMRNRQzEjUSEhMREzMSEhMRExATERMQEzQTNBISEzQSERMRExATERMQExETMxQzFDQTMxQ0FDIUMxI1EhITERMQExETEBMRExATERMQEzQTNBISEzMSNRISExETEBMRExATERMQExETEBI1EzQSNRISExATEBQQExETEBMREhETNBMQFDMUMxQQEzQSNRQQE9qRjxQzEjUUMxQzFBATERMzFBASERQQExAUEBM0EzQTERMzEhITEBQQExAUEBMQFDMSNRM0EjUUMxI1EzQSNRQQExATERMQFBASERMRExATERM0EjUSERQzEzQSEhMQFBATEBMRExASEhMQFBATMRY0EjUSERQQExATERMQFBATEBQQEzQSERM0EjUTERM0EjUTEBMADQU=
JgAqAY+QEjUSNRQzFDMSERQQFDMUDxUPFA8VDxQQEzMSNRUPFDMUDxUPFA8VDxQPFQ8UMxQzEjQVMxQzEjUUMxI1EhEVDxQPFQ8UDxUPFA8VDxQPFTISNRUPFDMUMxIRFQ8UDxUPFA8VEBMPFRAZChQyEjUSNRISFA8VDxQPFQ8UDxUPFA8VMhISFDMUMxQPFTISNRISFNmTjhQzEjUSNRQzFA8VDxQzFA8VDxQPFQ8UDxUyEjUUEBQzEhIUDxQPFQ8UDxUPFDMUMxQzFDMUMxQzEjUSNRQPFQ8UDxUPFA8UEBQPFQ8UDxUyEjUSEhQzFDMSERQQFA8UEBQPFQ8UDxQQFA8VMhQzEjUUEBQPFBAUDxQQFA8UEBQPFDMSEhQzFDMUDxQzEjQTEhQADQU=
JgAqAZCPEzMSNRI1FDMSEhQPEzQUEBQPFBATEBQQFDMRNhQPFDMUEBMQExEUDxMRFA8TNBQzFDMUMxQzEjUUMxQzFBAUDxQQExATERQPFBAUDxQQEzQTNBQPFDMTNBQQFA8TERMQExETEBMRFA8TERM0FDMUMxQPExETEBMRExATERQPExETNBMQFDMTNBQQEzQUMxQPFNqRjxQzFDMTNBM0FBAUDxM0FBATEBMRExATERQzFDMSERM0FBATEBQQExATERMQEzQTNBI1EzQTNBQzEzQTNBISFA8TERMQExETEBMRFA8TERQzEjUUDxM0FDMUEBQPExETEBMRFA8TERMSEhATNBQzEjMUERQQExAUEBMQExETEBkLEzQSERQzEzQTERM0FDISEhMADQU=
```

DarkStarSword, 2021-08-14:
I'm not a developer on this project, but have coded up an experimental script to convert the base64 encoded packets to waveforms and happened to notice your bug report. As you can see, the waveform is essentially the same on each of your captures, but because of the nature of sampling IR/RF, the exact timing can vary slightly which will lead to a different base64 encoded packet.

‾‾ and _ are hopefully fairly self explanatory: ‾‾ means a 433MHz signal is present the entirety of that sample, _ means it is not present at all. I've compressed the waveform by a factor of 12 compared to Broadlinks output, so / means a rising signal is present somewhere in the 12 samples that character represents, and \ means a falling signal was present (X would mean both are present and I've compressed too far). 433MHz signals almost always use OOK, so the signal is either present or not, very simple to work with.

P is a marker my script inserts where it thinks the preamble has ended and the real packet has begun. The preamble is a longer IR/RF blast that precedes the real signal to give the receiver time to adjust its gain so it can get a clear reading without background noise. My script lines up any resends of the packet on this marker to make it easy to confirm if the resends all match.

S is a marker my script inserts where it thinks a packet has ended (following pulse much shorter or longer than expected). If additional data follows (usually the sender repeating the signal, depending on which IR protocol is in use) it inserts a newline and looks for another preamble so that we can see the repeat lined up with the original.

... replaces the junk Broadlink inserts at the end of every IR capture that is not part of the actual signal.

```
  1    IR
                         ‾‾‾‾‾‾‾‾‾‾‾‾___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾_/‾\/‾_/‾_/\_/\____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾____/‾____/\____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/\_/\_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/\_/‾____/\_/\____/\____/\_/\____‾\___/‾\/‾S
      __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾_/\_/‾_/\_/\____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/\____/\____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/‾_/\_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾_/‾_/‾____/\_/\____/\____/\_/\____‾\___/‾\/‾S...
  2    IR
                         ‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾\/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾\___/\____/‾____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/\____/‾_/\_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾_/\____/\____‾\_/\____‾\___/‾\/‾S
      __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾_/‾\/‾\/‾_/‾_/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/\____/\____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/\____/‾_/‾_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/\_/\____/\____/\_/\____‾\___/‾\/‾S...
  3    IR
                         ‾‾‾‾‾‾‾‾‾‾‾‾___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾\/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/\____/\____/\____/\____/\_/\_‾\_‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾____/‾\/\_/\_/\_/\_/\_/\_/\_/\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾_/‾\___/\_/\____/\____/\_/\____‾\___/‾\/‾S
      __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾_/‾_/\____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾____/\____/‾____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/\_/‾_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾_/‾\___/‾_/\____/\____/\_/\____‾\___/‾\/‾S...
  4    IR
                         ‾‾‾‾‾‾‾‾‾‾‾‾\__________P‾\____‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾\/‾____/\____/\_/\____/\_‾\_‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾____/‾\___/\____/\____/\_/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾____/‾\/‾\/‾_/\_/\_/\_/\_/\_/\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/\_/‾____/\____/\_/\____‾\___/‾\/‾S
      __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾_/‾\/‾\___/‾\/‾\/‾_/\_/\_/\____/\____‾\_‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾____/‾\___/‾____/\____/\____/\____‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾____/‾\/‾\___/‾____/\_/\_/\_/\_/\_‾\_‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾_/‾_/‾____/\_/\____‾\____‾\/‾\___/‾\___/‾\/‾S...
  5    IR
                         ‾‾‾‾‾‾‾‾‾‾‾‾\__________P‾\____‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾\/‾\___/\____/\_/\____/\_/\_‾\_‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾____/\____/\____/‾____/\_/\_/\_/\_‾\_‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾____/‾\/‾\/‾\/‾_/‾_/\_/\_/\_/\_/\____‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\/‾\___/\____/\_/\____/\____‾\/‾S
      __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾_/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/\____/‾____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/‾\/‾_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾_/\____/\____/\_/\____‾\___/‾\/‾S...
  6    IR
[... cut ...] /‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/\____/\_/\_/\_/\_/\_/\_‾\_‾\/‾\/‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾_/‾\/\_/\____/\_/\____/\____‾\_‾\___/‾\___/‾\/‾S
      __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾_/‾_/\_/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾____/‾____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/‾_/‾_/\_/\_/\_/\_/\_/\_/\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾____/\_/\____/\____/\_/\____‾\___/‾\/‾S...
```

Zen3515, 2021-08-14:
> I'm not a developer on this project, but have coded up an experimental script to convert the base64 encoded packets to waveforms and happened to notice your bug report. As you can see, the waveform is essentially the same on each of your captures, but because of the nature of sampling IR/RF, the exact timing can vary slightly which will lead to a different base64 encoded packet.
> 
> ‾‾ and _ are hopefully fairly self explanatory: ‾‾ means a IR/RF signal is present the entirety of that sample, _ means it is not present at all. I've compressed the waveform by a factor of 12 compared to Broadlinks output, so / means a rising signal is present somewhere in the 12 samples that character represents, and \ means a falling signal was present (X would mean both are present and I've compressed too far).
> 
> P is a marker my script inserts where it thinks the preamble has ended and the real packet has begun. The preamble is a longer IR/RF blast that precedes the real signal to give the receiver time to adjust its gain so it can get a clear reading without background noise. My script lines up any resends of the packet on this marker to make it easy to confirm if the resends all match.
> 
> S is a marker my script inserts where it thinks a packet has ended (following pulse much shorter or longer than expected). If additional data follows (usually the sender repeating the signal, depending on which IR protocol is in use) it inserts a newline and looks for another preamble so that we can see the repeat lined up with the original.
> 
> ... replaces the junk Broadlink inserts at the end of every IR capture that is not part of the actual signal.
> 
> ```
>   1    IR
>                          ‾‾‾‾‾‾‾‾‾‾‾‾___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾_/‾\/‾_/‾_/\_/\____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾____/‾____/\____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/\_/\_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/\_/‾____/\_/\____/\____/\_/\____‾\___/‾\/‾S
>       __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾_/\_/‾_/\_/\____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/\____/\____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/‾_/\_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾_/‾_/‾____/\_/\____/\____/\_/\____‾\___/‾\/‾S...
>   2    IR
>                          ‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾\/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾\___/\____/‾____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/\____/‾_/\_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾_/\____/\____‾\_/\____‾\___/‾\/‾S
>       __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾_/‾\/‾\/‾_/‾_/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/\____/\____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/\____/‾_/‾_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/\_/\____/\____/\_/\____‾\___/‾\/‾S...
>   3    IR
>                          ‾‾‾‾‾‾‾‾‾‾‾‾___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾\/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/\____/\____/\____/\____/\_/\_‾\_‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾____/‾\/\_/\_/\_/\_/\_/\_/\_/\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾_/‾\___/\_/\____/\____/\_/\____‾\___/‾\/‾S
>       __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾_/‾_/\____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾____/\____/‾____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/\_/‾_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾_/‾\___/‾_/\____/\____/\_/\____‾\___/‾\/‾S...
>   4    IR
>                          ‾‾‾‾‾‾‾‾‾‾‾‾\__________P‾\____‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾\/‾____/\____/\_/\____/\_‾\_‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾____/‾\___/\____/\____/\_/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾____/‾\/‾\/‾_/\_/\_/\_/\_/\_/\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/\_/‾____/\____/\_/\____‾\___/‾\/‾S
>       __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾_/‾\/‾\___/‾\/‾\/‾_/\_/\_/\____/\____‾\_‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾____/‾\___/‾____/\____/\____/\____‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾____/‾\/‾\___/‾____/\_/\_/\_/\_/\_‾\_‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾_/‾_/‾____/\_/\____‾\____‾\/‾\___/‾\___/‾\/‾S...
>   5    IR
>                          ‾‾‾‾‾‾‾‾‾‾‾‾\__________P‾\____‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾\/‾\___/\____/\_/\____/\_/\_‾\_‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾____/\____/\____/‾____/\_/\_/\_/\_‾\_‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾____/‾\/‾\/‾\/‾_/‾_/\_/\_/\_/\_/\____‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\/‾\___/\____/\_/\____/\____‾\/‾S
>       __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾\/‾\/‾_/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/\____/‾____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/‾\/‾_/\_/\_/\_/\_/\_/\_‾\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾_/\____/\____/\_/\____‾\___/‾\/‾S...
>   6    IR
>                                                 P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾_/‾_/\_/\_/\____/\____/\_/\____‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾____/\____/‾____/\____/\____/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/\____/‾_/ [... cut ...]
>       __________________/‾‾‾‾‾‾‾‾‾‾‾\___________P‾\___/‾\___/‾\___/‾\___/‾\/‾\/‾\___/‾\/‾\/‾_/‾_/\_/‾____/\____/\_/\____‾\_‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\___/‾\___/‾____/‾____/\____/\____/\_/\_‾\_‾\/‾\/‾\/‾\/‾\/‾\/‾\___/‾\___/‾\/‾\___/‾\___/‾_/‾_/\_/\_/\_/\_/\_/\_/\_‾\___/‾\___/‾\___/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾\/‾____/\_/\____/\____/\_/\____‾\___/‾\/‾S...
> ```
> 
> I'll probably stick the script up in my junk code later, once it's a little less hardcoded.

This got me even more confused, if the payload itself is the same, then why mc AC choose to follow some of the learned signal.
If I transmit these codes, only some of them will actually turn off my ac.

DarkStarSword, 2021-08-15:
Can't be sure why that would be - perhaps the unit is quite sensitive to the precise timing?

Capturing a signal on the RF/IR receiver side will never be as precise as hooking up an oscilloscope / logic analyser to the transmitter, since the receivers lag a little behind the signal with inconsistent timing. Usually it's good enough, but not always. :shrug:

Zen3515, 2021-08-19:
I can confirm that all the commands I learned is actually the same.
I plot the rest of the signal to find out that they are not really change, just normal behavior.

For my problem, sometime only when the signal is shorter than the other, that's the one that's likely to not work as the repeat or the first signal is wrong.

closed 2021-08-19 by Zen3515