mjg59/python-broadlink #611: conn.recvfrom(2048)[0] socket.timeout: timed out

bytos, 2021-08-12:
Request for humidity sensor at regular intervals met with this error.

felipediel, 2021-10-17:
Hi. This is an environment issue. The new firmware requires access to the cloud. Check if the device has an internet connection, or if it has a blinking LED. If you blocked the device with a firewall, you will have to send watchdog packets regularly to prevent it from restarting. You can do it with the broadlink.ping() function. If you use Home Assistant, we send this packets automatically, so you don't need to care about this.

Closing, not something we can fix here :+1:

closed 2021-10-17 by felipediel

bytos, 2021-10-17:
The RM4 Mini is constantly reconnecting looking for Internet access, but I'm using the device with my own code and not allowing these devices to reach the Internet. Thanks.