mjg59/python-broadlink #617: Home Assistant 2021.9.6 | Switches | Entities unavailable

bigsister-git, 2021-09-26:
**Prerequisites**
The Remote is integrated and is displayed accordingly 
[image]

In switch.yaml the switches are defined as follows

```
  - platform: broadlink
    mac: 81:C6:B9:A7:DF:25
    switches:
      - name: Philips ON/OFF
        command_on: JgBUAFcaDh4ODw4PLCsODw8ODw4ODw8ODw4PDg8ODw8ODw4PHQ4OHg4PDQAKqFcbDh4ODw4PKi0ODw0QDg8ODw4PDw4PDg4PDw4ODw4PHQ8OHQ4PDgANBQAA
      - name: Philips Volume UP/DOWN
        command_on: JgBUAFYbDx0NEA0QLCoQDhEMDRANEA8ODhAMEA4QDRANEBsfDg8NEA4PDQAKpVcbDR8ODw4PKywNEA4PDg8NEA8ODw4RDA4PDw4PDhweDRAOEA4PDgANBQAA
        command_off: JgBUAFgZDR8NEA8ODR4dDw4OEQ0ODw8ODg8NEA4PDRAODw4PHB4NEA4QHAAKs1UdDxwODw4PDh4eDgwRDBENEA8ODg8NEA4PEgsPDg8OHR0QDQ8OHgANBQAA 
```
and so on...

**Actual results**
1. The Switches are displayed as unavailable
[image]
2. In Developer Tools the swithes are not created
[image]

**What I did**
1.  Delete the remote -> Restart -> Initialised
2. Even though I took the MAC address from the router itself I reverse the order of the bites

**Expected results**
The switches must work as designed

**OBS**
I have Broadlnk outlet, too. These are working perfectly. The only issues that I have are with switches that are defined based on IR / RF codes.

CarpeDiemRo, 2021-09-26:
@felipediel I have exactly the same problem.

felipediel, 2021-09-26:
Hi. Please triple check the MAC address.

MichielioZ, 2021-09-28:
I personally never used a switch.yaml file for anything (IR or RF) concerning the RM4 Pro I have.
In my case I learned the commands through Home Assistant, so they are stored by device name in the /.storage/broadlink_{mac} file. I could then use them directly in either Lovelace or Node-RED by using the device's name and turn_on/turn_off command.

In case that doesn't work for you it might be good to drop any special characters in names (I avoid those as the plague), so maybe try the name without the "/" betwoon ON and OFF ?!

felipediel, 2021-10-18:
Looks like the issue is solved. If the problem persists, please reopen it in Home Assistant's repository.

closed 2021-10-18 by felipediel