mjg59/python-broadlink #637: Help with install of pip3 install broadlink

vjmaz, 2021-11-01:
Hi team I went to run the pip3 Install broadlink ( and notice red writing along the way is this ok as when I get to 
command : python 
then 
import broadlink   
(it feels like its not right)   I'm no python user but have many Broadlink lights to install Team in HA  if someone could guide me please team Thank you 

➜  ~ python3
Python 3.9.5 (default, May 12 2021, 20:44:22)
[GCC 10.3.1 20210424] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'broadlink'
>>>

I have a txt output of the install nay suggestion's please

vjmaz, 2021-11-01:
[Install of pip3 insatll broadlink.txt](https://github.com/mjg59/python-broadlink/files/7452133/Install.of.pip3.insatll.broadlink.txt)

vjmaz, 2021-11-01:
could someone maybe provide a manual installation please

ghiki, 2021-11-03:
Same problem here with "cryptohraphy" while intalling "broadlink":
[image]

felipediel, 2021-11-04:
Hi. What are you trying to achieve? If you use Home Assistant, you don't need to install this module, it is updated automatically when updates for the Broadlink integration arrive.

If you are doing something different, you can try to install the dependencies that pip is asking for:
```
pip3 install --upgrade pip
pip3 install wheel
pip3 install setuptools
pip3 install setuptools-rust
```

ghiki, 2021-11-04:
> Hi. What are you trying to achieve? If you use Home Assistant, you don't need to install this module, it is updated automatically when updates for the Broadlink integration arrive.
>
> If you are doing [...]

The problem is, when I try to install "broadlink" through PIP, It tries to compile/install cryptography, but It fails since It doesn't find "rust" compiler.

I tried in Android with PyDroid (pip install broadlink, after upgrading PIP) and It does the same error. Apparently when It tries to install "rust", the versión that's installed is 0.1.1.

felipediel, 2021-11-04:
Did you try [this](https://github.com/mjg59/python-broadlink/issues/637#issuecomment-960356637)?

ghiki, 2021-11-04:
> Did you try [this](https://github.com/mjg59/python-broadlink/issues/637#issuecomment-960356637)? Actually, I think simply upgrading pip will do it.
> 
> ```
> pip3 install --upgrade pip
> pip3 install broadlink
> ```

Yep, It outputs "this package requieres rust >= 1.41.0" as you can see in the screenshot (It outputs when It tries to compile cryptography). If you have Android phone you can see by yourself in "PyDroid"

felipediel, 2021-11-04:
Did you upgrade pip first?

ghiki, 2021-11-04:
> Did you upgrade pip first? You only need to upgrade pip.

Yes, I upgraded first 
[image]

closed 2021-11-04 by felipediel

vjmaz, 2021-11-05:
So you asked what are you trying to achieve ( I have Broadlink device mainly IR & RF working well )  then I purchased these Broadlink Light bulbs which work well with Broadlink App but god help us trying to bring it into HA ... my god what a pain !!! 

Anyway I have 4 of them which I will try to resell and I'm going to stick with Shelly as they are truly the only IOT provider who care about Local setup all other brands I'm not touching no more or wasting time with Especially TUYA or related products ... I have also made a decision to not touch ZigBee as I have over 40 Light bulbs with WeMo but I have lost faith in any mesh system now as I'm sick of these drop outs .... also the limitation of only one HA server allowed ... All my other IOT or smart devices that are on wifi can work with my test HA and my Main HA Servers so i can actually do true tests when needed ...  I cant comment on ZWave devices but they are a mesh system which I'm not going to bother wifi or true cat5 connection only now ... This said I will add one more thing I love my Broadlink IR & RF devices they work really really well and they also can be run from multiple HA servers :) ... 
shame on the Broadlink Light bulbs they work really well in their Broadlink app 

I tried this below no success 
..................................
Hi. What are you trying to achieve? If you use Home Assistant, you don't need to install this module, it is updated automatically when updates for the Broadlink integration arrive.

If you are doing something different, you can try to install the dependencies that pip is asking for:

pip3 install --upgrade pip
pip3 install wheel
pip3 install setuptools

felipediel, 2021-11-05:
They will be supported in the Home Assistant, but we still need to write the code. Updating the library alone won't do it.