mjg59/python-broadlink #641: Cloud ping packets not working with RM4 pro with firmware 52079

trancefam, 2021-12-30:
Hello, I hope this is the correct repo for this. I use Home Assistant and have 2 different broadlinks:

[image]

[image]

My issue is with the RM4 pro -- it disconnects and reconnects every 3 minutes to my unifi AP. All outbound access for both broadlinks are blocked at my firewall. This seems directly related to [this prior issue which aimed to add support for sending cloud ping packets](https://github.com/mjg59/python-broadlink/issues/458). After reviewing the logs from pfsense, I can see that both of the cloud ping packets are being sent automatically by home assistant. The RM3 mini stays connected; however, the RM4 pro (with newer firmware) does not.

Has anyone run into this issue? Has the cloud ping packet changed? Any help is greatly appreciated.

PFSense logs for RM4 pro below (10.0.1.50=Home assistant, 10.30.4.9=RM4 Pro):

| timestamp                | pf\_ip\_source\_ip | pf\_ip\_destination\_ip                                                 | message | pf\_ipv4\_protocol | pf\_udp\_destination\_port | pf\_action |
| ------------------------ | ------------------ | ----------------------------------------------------------------------- | ------- | ------------------ | -------------------------- | ---------- |
| 2021-12-30T02:18:50.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:19:50.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:**19:51**.000Z |                    | dhcpd\[70790\]: DHCPACK on 10.30.4.9 to 24:df:a7:f0:ea:e4 via igb2.4000 |         |                    |                            |
| 2021-12-30T02:20:24.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:20:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:21:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:22:24.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:**22:50**.000Z |                    | dhcpd\[70790\]: DHCPACK on 10.30.4.9 to 24:df:a7:f0:ea:e4 via igb2.4000 |         |                    |                            |
| 2021-12-30T02:22:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:23:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:24:23.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:24:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:**25:49**.000Z |                    | dhcpd\[70790\]: DHCPACK on 10.30.4.9 to 24:df:a7:f0:ea:e4 via igb2.4000 |         |                    |                            |
| 2021-12-30T02:25:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:26:23.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:26:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:27:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:28:24.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:**28:53**.000Z |                    | dhcpd\[70790\]: DHCPACK on 10.30.4.9 to 24:df:a7:f0:ea:e4 via igb2.4000 |         |                    |                            |
| 2021-12-30T02:28:53.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:29:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:30:24.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:30:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:**31:52**.000Z |                    | dhcpd\[70790\]: DHCPACK on 10.30.4.9 to 24:df:a7:f0:ea:e4 via igb2.4000 |         |                    |                            |
| 2021-12-30T02:31:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:32:23.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:32:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:33:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:34:23.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:**34:52**.000Z |                    | dhcpd\[70790\]: DHCPACK on 10.30.4.9 to 24:df:a7:f0:ea:e4 via igb2.4000 |         |                    |                            |
| 2021-12-30T02:34:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:35:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:36:24.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:36:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:37:55.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:**37:56**.000Z |                    | dhcpd\[70790\]: DHCPACK on 10.30.4.9 to 24:df:a7:f0:ea:e4 via igb2.4000 |         |                    |                            |
| 2021-12-30T02:38:24.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
[... cut ...]
| 2021-12-30T02:46:24.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:46:57.000Z | 10.0.1.50          | 10.30.4.9                                                               |         | udp                | 80                         | pass       |
| 2021-12-30T02:**46:58**.000Z |                    | dhcpd\[70790\]: DHCPACK on 10.30.4.9 to 24:df:a7:f0:ea:e4 via igb2.4000 |         |                    |                            |

felipediel, 2022-02-03:
No idea, but thanks for reporting this. Could you allow internet access and sniff the new ping packet with your firewall?

mainTAP, 2022-03-18:
RM4C mini also doesn't work with the current ping packet.
Looking at the heartbeat, there is UDP communication towards 13.52.64.71:1812 ( 5278v2main.broadlink.com.cn  and also 5278v2backup.broadlink.com.cn when the first is not responding )

The UDP payload is changing :
```
0000   d2 7f df 02 01 00 8c 00 49 5b da 09 99 16 4a 04   ........I[....J.
0010   e9 e4 fe de 1b 82 da 09 48 5b da 0b 13 c7 e5 e4   ........H[......
0020   da 43 5a 59 22 8e 83 8f 15 07 94 63 3b 5d 00 6d   .CZY"......c;].m
0030   99 40 02 63 59 a4 9a e8 c8 98 e6 3e bd 25 17 d7   .@.cY......>.%..
0040   dc 51 98 a5 6a fa fb 95 6f d4 00 13 23 b7 db ac   .Q..j...o...#...
0050   7d 69 9c 2c 36 da 01 61 e3 7e 2a 85 c6 e9 fe 05   }i.,6..a.~*.....
0060   a4 b9 9c fd 14 de 3f cc 31 81 2c c2 9a ba 33 4f   ......?.1.,...3O
0070   87 f8 ed ec df 7f 05 66 22 28 51 7f 57 6d 7b bf   .......f"(Q.Wm{.
0080   a2 f9 33 3a 69 ad 1e 5a 53 d3 f9 c4               ..3:i..ZS...

0000   d2 7f df 02 01 00 8c 00 1a a8 dd 09 f2 e5 4d 04   ..............M.
0010   b8 4f 74 8e 48 71 dd 09 1b a8 dd 0b 40 34 e2 e4   .Ot.Hq......@4..
0020   89 b0 5d 59 71 7d 84 8f 46 f4 93 63 68 ae 07 6d   ..]Yq}..F..ch..m
0030   ca b3 05 63 0a 57 9d e8 9b 6b e1 3e 8e 21 9b 1c   ...c.W...k.>.!..
0040   d4 33 fc cc 58 48 15 ec 30 22 c0 4e c4 27 6d df   .3..XH..0".N.'m.
0050   fc 05 7e 03 d0 3e d0 a7 09 7b ba 32 e0 57 72 37   ..~..>...{.2.Wr7
0060   d0 51 78 46 db df 2d 49 8f 48 22 9b b0 e0 99 98   .QxF..-I.H".....
0070   ea ef 92 e0 a9 4a f0 a8 0b ef 1e 6b a6 00 5c d7   .....J.....k..\.
0080   74 82 a1 19 0d 96 8c 99 14 87 2a b5               t.........*.
```

felipediel, 2022-03-21:
It changes because the message is encrypted. I can't devote myself to this right now, but if anyone is interested, [these tools](https://github.com/felipediel/broadlink-hacktools) can be used to decrypt the communication.

Write, 2022-05-15:
> It changes because the message is encrypted. I can't devote myself to this right now, but if anyone is interested, [these tools](https://github.com/felipediel/broadlink-hacktools) can be used to decrypt the communication.

Hi, I tried to use your tools but can't get to export to bin and correctly use broadlink-hacktools to read packets data.
However maybe giving you my .pcap could help you ? Where 192.168.1.112 is my broadlink RM4C Mini

https://cloud.socialspill.com/f/598060

Write, 2022-06-05:
Would love to help more on this but I really don't know how to use my .cap binary files with broadlink-hacktools

jiehanzheng, 2022-06-29:
Does anyone happen to have a copy of one of the older firmwares that we could downgrade to?

Running into the same issue :(

Write, 2022-06-29:
You can downgrade Broadlink devices ? I would love to do that too honestely it's so annoying.

jiehanzheng, 2022-07-13:
> You can downgrade Broadlink devices ? I would love to do that too honestely it's so annoying.

I managed to get a previous version firmware URL from Broadlink support, and tried to downgrade 3 times, and all 3 attempts failed.

Feel free to message them on Amazon or via email and see if it works for you.  Hope you have better luck :)

xxyyttxx, 2022-09-29:
> The UDP payload is changing :
> 
> ```
> 0000   d2 7f df 02 01 00 8c 00 49 5b da 09 99 16 4a 04   ........I[....J.
> ```

I captured same payloads which start with `d2 7f df 02 01 00 8c/9c`.
The old-style payloads should start with `5a a5 aa 55 5a a5 aa 55`.
It seems that broadlink is using new way to encrypt payloads for its cloud ping.

Write, 2022-09-29:
> I captured same payloads which start with `d2 7f df 02 01 00 8c/9c`. The old-style payloads should start with `5a a5 aa 55 5a a5 aa 55`. It seems that broadlink is using new way to encrypt payloads [...]

Did you used https://github.com/felipediel/broadlink-hacktools to actually decrypt it ? Otherwise it’s normal it’s always different since it’s encrypted.

xxyyttxx, 2022-10-11:
> > > The UDP payload is changing :
> > > ```
> > > 0000   d2 7f df 02 01 00 8c 00 49 5b da 09 99 16 4a 04   ........I[....J.
> > > ```
> > 
> > 
> > I captured same payloads which start with `d2 7f df 02 01 00 8c/9c`. The old-style payloads should start with `5a a5 aa 55 5a a5 aa 55`. It seems that broadlink is using new way to encrypt payloads for its cloud ping.
> 
> Did you use https://github.com/felipediel/broadlink-hacktools to actually decrypt it ? Otherwise it’s normal it’s always different since it’s encrypted.

I have tried, and it output: 

```

+---------------------------------------------------------------------------------------------------------------------------------------------------------------------+----------+-------+-------------+---------+---------+-------------------+-------------+------------------+-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------+---------+
|                                                                               Header                                                                                | Checksum | Error | Device type | Command | Counter |    MAC address    |  Device ID  | Payload checksum |                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    Encrypted payload                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    | Payload |
+---------------------------------------------------------------------------------------------------------------------------------------------------------------------+----------+-------+-------------+---------+---------+-------------------+-------------+------------------+-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------+---------+
| +-------------------------+-------------+-------+---------+---------+-------+-----------------+--------------+-------+-------------+----------------+-------------+ |  24 2b   | 33 06 |    bb 68    |  cb a0  |  7d f8  | de 38 da 61 92 49 | 54 97 45 c4 |      cc e8       | a5 39 1a 84 44 8d f4 bb 8e f9 64 fc 69 bf b4 bd 20 01 02 b9 81 9e e1 b6 8a da a5 68 a7 83 6c eb f1 66 c8 b5 03 78 b2 f2 fd 0f 4b 77 b0 9d ce 88 74 e0 2c f9 78 e2 d4 dd 59 a1 bc e9 f2 6d 18 7f 54 41 23 16 81 c8 5f 10 d5 8f be ba f1 15 20 ef 6b d0 d3 1f 0e 01 35 39 20 af d5 81 e6 30 86 e4 09 6c 10 59 05 88 a6 01 74 b7 32 00 07 df 1a 2f fa 26 88 5e fa 5b 98 36 c5 27 a9 0c e0 df 8f 83 70 bf 2c 8b 9b 7d 0a 21 5d ca a2 f2 d6 8f 2a ea c9 58 e4 92 7c 81 8a ec 46 fb 37 40 d3 0b 71 2d 71 95 7d 9c 95 49 1b f3 37 95 fa 59 41 3b af 06 a1 b0 d7 e3 13 46 27 51 27 fd 2d db 72 68 30 99 53 48 6f da d7 fd 69 2f 34 ac de 2e 1c aa f3 53 9a a8 35 04 a1 12 2e 3c b6 ec 9b 2e 18 bf 8d e4 20 9d 87 4f 51 c5 c2 11 0c b7 01 c6 49 a8 0b ce 56 88 5b ac 9b b0 fc cf 1f 71 93 75 92 c1 22 22 a3 6c 96 3b 32 eb 65 55 b7 1f 6a 2a 6d 5a ec 73 63 d1 55 a3 e4 ed de 8a 91 2f 9b c4 7 [... cut ...] +----+----+-----+-----+-----+-----+-----+-----+----+-----+----+----+-----+----+---+-----+-----+----+----+-----+----+----+----+-----+---+----+-----+-----+-----+----+----+----+----+---+-----+

```

I presume that the hacktools cannot deal with it.
what's more, I found that 229 out of 252 payloads in [the repo](https://github.com/felipediel/broadlink-hacktools/tree/master/examples/samples) are start with `5aa5`, and others are start with `0000`, which also support my view.

What about your opinion?

Write, 2022-11-20:
@felipediel Please help us, what whould you need ? 😭

felipediel, 2022-11-20:
Indeed, the hacktools cannot deal with it, the packets have a different structure. The way is to reverse engineer the new firmware and try to understand the structure of the packets and the type of encryption used. This task takes some time and I cannot dedicate myself to it. If anyone can replicate a working watchdog packet, I'll be happy to turn it into production code.

pablolucas890, 2023-01-23:
Same problem for RM4 MINI with V52078 / V52079 Firmware Version =/

andreluis034, 2023-05-23:
I bought two rm4c mini both have the firmware 52079 and they periodically ping a http server at http://25913service.ibroadlink.com/devactiva/v2/devactiva

Unfortunately the content is encrypted with what I assume to be also AES-CBC with a constant IV. This is the captured packet:

[image]

The reason I believe they are AES-CBC with a constant IV and key, is that both the initial request only differ at the end, probably due to a timestamp. Logging the requests I get:

```
::ffff:192.168.99.36 3fe609fc32211da63ec868421ae2d663fea27b780d927f92a3bd5a4e0614b364e764194774086e7f1b09e442e2c5a3aecb9d876116592d725ebe738d5c668bb6c6dac146361abd51d2f9f8dfec591eeb0dd7340986707f30b3285a8881398c70eaf57c4dacf260fdac8249e2a57221749f7e46fe6fc5d55c9baa1ce8a5539859b984efa3b700d4825631bf4da4df71a134af6299c4d51c606593fa0fb2b2405e06ef4188df291a757a5db5c673a718a0c5b47b6d447676180f16f2b0fb8089e730fc4ee3f5941ddffd82bb371e9c97a4e9fd64591151d4d1dafba4c1ffee4acb189691d7924400b080283de56f2202dd245af0949edab4b2d98e74156ea2ab5a2b394d94521b5c54773d5bc2498f81579f05c664b20fff31188098ee28d02868ed156b6f0337b1cd0e8346e11f7f0801
::ffff:192.168.99.36 3fe609fc32211da63ec868421ae2d663fea27b780d927f92a3bd5a4e0614b364e764194774086e7f1b09e442e2c5a3aecb9d876116592d725ebe738d5c668bb6c6dac146361abd51d2f9f8dfec591eeb0dd7340986707f30b3285a8881398c70eaf57c4dacf260fdac8249e2a57221749f7e46fe6fc5d55c9baa1ce8a5539859b984efa3b700d4825631bf4da4df71a134af6299c4d51c606593fa0fb2b2405e06ef4188df291a757a5db5c673a718a0c5b47b6d447676180f16f2b0fb8089e730fc4ee3f5941ddffd82bb371e9c97a4e9fd64591151d4d1dafba4c1ffee4acb189691d7924400b080283de56f2202dd245af0949edab4b2d98e74156ea2ab5a2b394d94521b5c54773d5bc2498f81579f05c664b20fff31188098ee28d028683d22a9ead695dff2f79e5c40c28021a7
```
I have tried to replay the answer from the server, but after a few seconds, the broadlink device ignores the answer, probably due to it using the timestamp fields as a nonce.

I do not mind trying to reverse engineer the firmware, but before I take apart my units to try and dump the firmware, do you @felipediel, have any way easy way to get the firmware (some download link, or whatever) ?

felipediel, 2023-05-24:
Hi @andreluis034. I know a guy who asked their support and they sent him a download link a while ago, but it was the previous version. I think it's worth asking them.

Two other approaches that you can try:

1. Spoofing the response from http://25913service.ibroadlink.com/devactiva/v2/devactiva with an empty packet, just to see if it cools down the watchdog timer.

2. Reseting the device and using this module to provision it. It needs to be removed from the official app before, and never added again.

Write, 2023-05-24:
I removed my RM 4c Mini from Broadlink App, and then reset my RM 4c, and provisioned it from my mac, sending the cloud ping to no avail, it still disconnect after 3 min.
Guess there's no other way than to find the latest firmware and reverse engineer it

andreluis034, 2023-05-24:
> Hi @andreluis034. I know a guy who asked their support and they sent him a download link a while ago, but it was the previous version. I think it's worth asking them.

I will give this one a try. If it does not work, I will try to reverse the android app to see if I can find the functionality to download the update. 

> 1. Spoofing the response from http://25913service.ibroadlink.com/devactiva/v2/devactiva with an empty packet, just to see if it cools down the watchdog timer.

Unfortunately it does not... What happens instead, is that the broadlink device starts making even more requests, one after the another. Under normal circumstances it pings the cloud every ~15s.
 
> 2. Reseting the device and using this module to provision it. It needs to be removed from the official app before, and never added again.

I never registered my device with the cloud. The only thing I have done, was take it out of the box and use `broadlink.setup` to assign it to an AP I have. The only thing that could have happened and still have not checked, was the devices performing some sort of cloud registration, before I could block them on my router's firewall.

Write, 2023-05-24:
Good news ! I've managed to downgrade my RM4C Mini, by combining the URL from this dead link : https://www.reddit.com/r/broadlink/comments/hmqh9c/help_me_update_firmware_rm4_mini_v52075/

And using the IP from this thread : https://www.reddit.com/r/homebridge/comments/1181ex4/caution_to_people_using_broadlink_rm_plugin/j9eqq0j/

Which made me able to download the v52075 firmware by combining the new host with the path for 52075 : http://124.71.130.238/RMBL4-52075-8710-BL-FC-2900.upd.bin

The firmware is supposedly for RM4 Mini, but I successfully downgraded my RM4c Mini to 52075.

Write, 2023-05-29:
Hello everyone,

After more testing, it seems that firmware 52075 makes cloud ping working again. 

I certainly made a mistake sending the cloud ping myself.
I wanted to downgrade yet again the firmware, as I've found new urls, I noticed the broadlink RM4c that I've downgraded to firmware 52075 no longer disconnect from the wifi (!). It was linked to HomeAssistant, which sends the fake cloud ping itself to the RM4c.

To be sure, I wanted to downgrade my other RM4c which still were on 52079.
Before I downgraded the 2nd one, I could see it disconnect every 3 minutes, I've downgraded it to 52075 too, then blocked Internet again, and added it to HomeAssistant. It never disconnected from the Wi-Fi again.

Can you guys confirm that firmware 52075 work fine with cloud ping / homeassistant ?

Here's the two older firmware (v52071 and v52074) urls I've found for the RM4c by scraping their website :
http://124.71.130.238/RMBL4-52071-8710-BL-FC-2626.upd.bin
http://124.71.130.238/RMBL4-52074-8710-BL-FC-2759.upd.bin

pablolucas890, 2023-05-31:
> Good news ! I've managed to downgrade my RM4C Mini, by combining the URL from this dead link : https://www.reddit.com/r/broadlink/comments/hmqh9c/help_me_update_firmware_rm4_mini_v52075/fy7tb9x/
> 
> And using the IP from this thread : https://www.reddit.com/r/homebridge/comments/1181ex4/caution_to_people_using_broadlink_rm_plugin/j9eqq0j/
> 
> Which made me able to download the v52075 firmware by combining the new host with the path for 52075 : http://124.71.130.238/RMBL4-52075-8710-BL-FC-2900.upd.bin
> 
> The firmware is supposedly for RM4 Mini, but I successfully downgraded my RM4c Mini to 52075.
> 
> **However** I can confirm the cloud ping method still doesn't work on this firmware. But hopefully @andreluis034 can help reverse engineer it ! And of course, the firmware in encrypted.
> 
> Binvis.io result on file entropy: [image]

Hello my friend, I receive your response on e-mail and I'll try to do the downgrade for broadlink RM4MINI and RM4PRO, you can help me on this? Do you use a specific program to do the firmaware downgrade?

VanQz, 2024-12-08:
Anyone solved this issue? im having this issue suddenly now.....

Write, 2024-12-08:
> Anyone solved this issue? im having this issue suddenly now.....

No.