mjg59/python-broadlink #642: broadlink_cli 0.18.0 won't learn RF codes

mjeshurun, 2022-01-02:
Hello, 
Thank you for creating this tool. 
Last year, I ran broadlink_cli on MacOS to learn RF remote codes with the RM4 Pro.
Yesterday I tried to learn new codes using the latest 0.18.0 version of broadlink_cli, but I was not able to capture new codes.
I tried running the command:
`python3 broadlink_cli --device "0x61a2 192.168.1.6 xxxxxxxxxx" --rfscanlearn` (substituting xxxxxxxxxxx with my RM4 Pro's MAC address).
The scan and learn process is initiated, and I'm able to capture my remote's RF frequency, but at stage 2 (actually learning the code), the process does not capture any remote codes and ultimately the process quits after a minute or so of inactivity.
Any one else has this issue? 
Any known fix? 
Hope someone can help.

MichielioZ, 2022-01-21:
Just to be sure: have you also tested with RF devices that worked before or only with new devices ?
(The RM4 Pro will not work with all RF devices)
Also if it worked before maybe try an older release from the releases page ?

mjeshurun, 2022-01-21:
@MichielioZ 
I tested with the same RF remotes I was able to learn in the past, but their remote codes also weren't learned.
OK, I'll try again with an older release version. I'll let you know if that solves the issue.
Thanks for the help.

mjeshurun, 2022-01-26:
Reverting back to version 0.15.0 fixed the RF learning issue and I was able to learn new remote codes.

closed 2022-01-26 by mjeshurun