mjg59/python-broadlink #643: RF Code learning process failed at "second button pres"

nmathey, 2022-01-03:
Hi there,

Trying to get RF Code from my Fan/Light remote: while the first long press button is working, the second press don't get anything and learning process fails.
Got it working though BroadLink app, where even if the second press fails, the button has been registered and I can use it.
Is there any way to get RF Code directly from the first long press of the learning process?

Cheers,
Nick

mjeshurun, 2022-01-26:
I had the same issue with version 0.18.0. 
I reverted back to version 0.15.0 and that fixed the RF learning issue and I was able to learn new remote codes.

felipediel, 2022-03-21:
> Hi there,
>
> Trying to get RF Code from my Fan/Light remote: while the first long press button is working, the second press don't get anything and learning process fails. Got it working though [...]

It will be possible to skip step 1 by specifying a frequency after https://github.com/mjg59/python-broadlink/pull/613:

```
broadlink_cli --device @BEDROOM.device --rflearn --radiofrequency 433.92
```

> I had the same issue with version 0.18.0. I reverted back to version 0.15.0 and that fixed the RF learning issue and I was able to learn new remote codes.

Are you sure? I don't remember any significant changes to the learning process.

mjeshurun, 2022-03-21:
> Are you sure? I don't remember any significant changes to the learning process.

When I tried to learn new RF codes in January using version 0.18.0, it failed. The only way I found to make it work was to revert back to 0.15.0. 
However, since you asked if I was sure, I just double checked version 0.18.0 and 0.18.1, and to my surprise, I was able to use them to learn new codes. 
So, I'm not sure what caused the problem in January.

felipediel, 2022-03-21:
Thanks for reporting back. I think this issue can be closed. If the problem persists, please reopen.

closed 2022-03-21 by felipediel