mjg59/python-broadlink #645: Add support for FastCon meshed devices

madmat777, 2022-01-06:
Broadlink devices use FastCon connections to create a mesh WiFi where they connect to each other instead of to the router/AP.

When the following code is executed

`import broadlink
for dev in broadlink.xdiscover():
   print(dev)` 

the result is only the devices that are connected to the router. The devices that are meshed do not show.

It must be possible to "look through" as both the Broasdlink phone App and Alexa can control all devices.

I've reviewed the FastCon SDK but it is beyond me on how this is possible - does not look easy.
https://docs.ibroadlink.com/public/appsdk_en/appsdk_08/

The interim solution is to put all devices into a network segment that does not have access to the internet. That forces them all to connect to the router (don't ask me why). The con with this is that Alex can not access the devices but the phone App can.

madmat777, 2022-01-06:
Not sure why the code is all on one line. Should be:

import broadlink
for dev in broadlink.xdiscover():
   print(dev)

felipediel, 2022-03-19:
Hi @madmat777. I opened a PR to add FastCon support, but I don't have the device to test. Could you help me?

madmat777, 2022-03-20:
Yes, for sure. I have them blocked from the internet at the moment but I can release the block and see how it goes.

madmat777, 2022-03-20:
@felipediel (in case you did not see my reply)

felipediel, 2022-03-20:
No need to allow internet access.