mjg59/python-broadlink #646: Support for Broadlink RM4 pro 0x5213

chpkt, 2022-01-16:
Please add support for a Broadlink RM4 pro, code `0x5213`

thanks!

jvcsw, 2022-01-20:
Hi,

how to find the code 0x5213 in the device?

I have the same problem, but I don't know if I have this exact code.

Thanks

chpkt, 2022-01-21:
add your Broadlink device to HA, after you got the "Device not supported" error, go to HA logs and you should find a similar error:
```
Unsupported device: 0x5213. If it worked before, please open an issue at https://github.com/home-assistant/core/issues
09:17:03 – (ERROR) Broadlink
```

jvcsw, 2022-01-21:
Yes, same problem here. 

Big disappointment. I really wanted to configure the broadlink remote control and the devices, this was not what I expected.

felipediel, 2022-01-23:
Added. It comes in the next release. Thank you!

closed 2022-01-23 by felipediel

jvcsw, 2022-01-24:
In the next release of Home Assistant or python-broadlink? Or both?

Thank you very much!

felipediel, 2022-01-24:
Both

jvcsw, 2022-02-03:
Hi @felipediel,

this message appears in the last version 2022.2 again:

"Unsupported device: 0x5213"

It doesn't work in the new version as you commented. 

Thanks

felipediel, 2022-02-03:
Hi. This device will be released in version 0.18.1 of this module, only then will we be able to create an update for Home Assistant.

jvcsw, 2022-02-03:
Ok, but, do you know when this will be? 
Because if it doesn't work for me I have to return the Broadlink product.

felipediel, 2022-02-03:
Soon