mjg59/python-broadlink #647: Add Support for S3 Hub 0xa59c

brianyulke, 2022-01-17:
Please add support for the S3 hub - 0xa59c.  This hub accompanies the Smart Light Switch TC3 (in my case, the TC3-US-1).

stevendodd, 2022-01-28:
+1 - I have the LC1 lights; is there anyway I can help I have basic python skills

nyulawbrian, 2022-01-28:
I've mentioned this in the Home Assistant feature request post I made about adding this same support -- I'm happy to work on this request myself with a little guidance from someone who has done this before.

stevendodd, 2022-01-29:
> +1 - I have the LC1 lights; is there anyway I can help I have basic python skills

`Python 3.6.0 (v3.6.0:41df79263a11, Dec 22 2016, 17:23:13) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
>>> device = broadlink.hello('192.168.1.99')
>>> device.auth()
True
>>> device.get_type()
'S3'
>>> device.get_state()
{'pwr': 0, 'indicator': 1, 'maxworktime': 0, 'childlock': 0}`

Where get_state() is more or less a copy of what is in light.py; I'm at a loss now as there is no documentation from broad link and I have no idea how to toggle sub devices or even list them.. This document gives me some idea, but nothing in terms of how to send the data to the hub  [https://docs.ibroadlink.com/public/appsdk_en/subdevice_api/](url)

stevendodd, 2022-01-31:
Here is the implementation hub.py

`"""Support for hubs."""
import struct
import enum
import json

from . import exceptions as e
from .device import Device

class s3(Device):
    """Controls a Broadlink S3."""

    TYPE = "S3"
    
    def print_payload(self) -> None:
        hex_string = "80958b19c3cdbbf4e44a07285a1f64a860dda677568d5369134ebef0345244d55dd827fdf7227d6424c73159bc311d0a336eec48a57830d4ccec3ebd06c176a8"  
        payload = self.decrypt(bytearray.fromhex(hex_string))
        js_len = struct.unpack_from("<I", payload, 0x08)[0]
        print(json.loads(payload[0x0C : 0x0C + js_len]))
        
        hex_string = "1dd713e6bf12fee9a7e98a29148b1f4ba569e3106ead7fb38214fa66584a3c4e98669e027ba94df43b8015719ee1b0feb8d83f7d14b28170357094e283024bc68eddfceca43e6376e5ad3a713ad04714"
        payload = self.decrypt(bytearray.fromhex(hex_string))
        js_len = struct.unpack_from("<I", payload, 0x08)[0]
        print(json.loads(payload[0x0C : 0x0C + js_len]))
    
    def set_state(
        self,
        did: str = None,
        pwr1: bool = None,
        pwr2: bool = None,
        pwr3: bool = None,
    ) -> dict:
        """Set the power state of the device."""
        state = {}
        if did is not None:
            state["did"] = did
        if pwr1 is not None:
            state["pwr1"] = int(bool(pwr1))
        if pwr2 is not None:
            state["pwr2"] = int(bool(pwr2))
        if pwr3 is not None:
            state["pwr1"] = int(bool(pwr3))

        packet = self._encode(2, state)
        response = self.send_packet(0x6A, packet)
        e.check_error(response[0x22:0x24])
        return self._decode(response)

    
    def _encode(self, flag: int, state: dict) -> bytes:
        """Encode a JSON packet."""
        # flag: 1 for reading, 2 for writing.
        packet = bytearray(12)
        data = json.dumps(state, separators=(",", ":")).encode()
        struct.pack_into("<HHHBBI", packet, 0, 0xA5A5, 0x5A5A, 0, flag, 0x0B, len(data))
        packet.extend(data)
        checksum = sum(packet, 0xBEAF) & 0xFFFF
        packet[0x04:0x06] = checksum.to_bytes(2, "little")
        return packet
    
    def _decode(self, response: bytes) -> dict:
        """Decode a JSON packet."""
        payload = self.decrypt(response[0x38:])
        js_len = struct.unpack_from("<I", payload, 0x08)[0]
        state = json.loads(payload[0x0C : 0x0C + js_len])
        return state`

stevendodd, 2022-02-06:
https://hub.docker.com/r/stevendodd/s3-rest-api

dj-fiorex, 2022-02-09:
Hello guys, i tried the fork of stevendodd, i also added all the files needed to turn this to a Home assistant addons, and it work well. I can connect to my Broadlink S3 Hub and get the actual state of the connected button Broadlink SR3-4KEY. My quesion is how can i listen to button click on the connected button? i'm a software engineer so i can write the code my self and help this library.

stevendodd, 2022-02-09:
Can you please share the output from device.get_state(did)

dj-fiorex, 2022-02-09:
Yes of course
```
{
"datatype": "fw_snapshot_v1",
"heartbeat": 0,
"lb_online1": 1,
"lowbattery": 0,
"scenarioswitch_1": 0,
"scenarioswitch_2": 1,
"scenarioswitch_3": 0,
"scenarioswitch_4": 0
}
```

this is a sample when the last button clicked was 2

stevendodd, 2022-02-09:
Does it stay like this until another button is pressed? What did you actually get working in home assistant have you also got light switches or other sub devices?

stevendodd, 2022-02-09:
And if you press button two again does it turn it off?

dj-fiorex, 2022-02-09:
Oh yes sorry, so
- it stays like this until another button is pressed,  i think because it is a battery device so to not drawn the battery
- no the button doesn't change the state
i started from the example addon on the ha github repo and then integrate it with your docker file, so i can run the s3 rest api server directly inside the addon (i use HA OS)

stevendodd, 2022-02-09:
Does that mean that if you press button to for a second time the scenario that you've hooked up in the broadlink app doesn't play for the second time?

stevendodd, 2022-02-09:
There must be some way to reset the button. I could extend my code so that you can turn the buttons on and off from the python libraries and the rest API. Then your logic would be something like if button state equals 1 do something; then sit button state equal to 0

stevendodd, 2022-02-09:
Could you also post the output from when you initialised the hub on the first request to the rest API

stevendodd, 2022-02-09:
The documentation from broadlink as always is pretty weak; I suspect you have not got any routines or triggers set up in the app and the button stays pressed. If you triggered a function in the app I believe the button will probably be reset to 0 after the function has triggered. Could you please test this theory for me.

dj-fiorex, 2022-02-10:
I already tested this, in fact i added push notification on the app settings, so if i click button 1 i will receive a notification on the phone.
But after the notification the json response is the same

stevendodd, 2022-02-10:
If you press button one twice in a row do you get a second notification?

stevendodd, 2022-02-10:
What happens if you press button one walk away for 5-minutes and then press button one again do you get two notifications

stevendodd, 2022-02-10:
Sorry to be a pain with all the questions, but once you click the 'I know it' on the notification can you check the json again to see if the button state is reset to 0. If it is not then the switch is sending a message to the hub rather than the hub polling the switch..  As I said above I can update the Python libraries to try and set the button state on or off however it may not trigger a routine in the broadlink app. Let's try it and see... I'll update the code in a bit and let you know

stevendodd, 2022-02-10:
> Could you also post the output from when you initialised the hub on the first request to the rest API

dj-fiorex, 2022-02-10:
No, don't worry for the questions, we are trying to help each other, like i said before the button state not reset to 0.
I used Wireshark to see the comunication of the hub, and every time i click the button, the S3 sends an udp message to the broadlink server i think.
 
this is the output of homeassistant.local:5000/?hub=192.168.1.74
`[{'did': '00000000000000000000ec0bae371ee1', 'pid': '00000000000000000000000048650000', 'name': '', 'offline': 0}]`

stevendodd, 2022-02-10:
Your switch must send a request to the hub when the button is pushed, the app can then monitor some sort of state on the hub in order to trigger a routine.

Try this: https://hub.docker.com/r/stevendodd/s3-rest-switch-api you should be able to set the state of the buttons on your switch with a POST to: homeassistant.local:5000/00000000000000000000ec0bae371ee1/1 for example. The post body needs to contain - {"active": "true"} to turn the button on or {"active": "false"}' for off.

I would be interested to know if you are able to turn the button on/off and when you turn it on is the routine triggered

dj-fiorex, 2022-02-10:
Just tested this, but not with your premade docker image, but with your repo,  i made a POST request with raw body 
```
{
    "active":true
}
```

but i got a network timeout for no response

```
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python3.9/site-packages/flask/app.py", line 1518, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python3.9/site-packages/flask/app.py", line 1516, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python3.9/site-packages/flask/app.py", line 1502, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/test/s3_rest.py", line 105, in dynamic
    return handle_request(request, did, int(gang))
  File "/test/s3_rest.py", line 38, in handle_request
    return create_resp(device.set_state(did, pwr[1], pwr[2], pwr[3]), gang)
  File "/test/broadlink/hub.py", line 66, in set_state
    response = self.send_packet(0x6A, packet)
  File "/test/broadlink/device.py", line 309, in send_packet
    raise e.NetworkTimeoutError(
broadlink.exceptions.NetworkTimeoutError: [Errno -4000] Network timeout: No response received within 10s
```

stevendodd, 2022-02-10:
The alternate image has new code to support your switch 

```
    def set_state(
        self,
        did: str = None,
        pwr1: bool = None,
        pwr2: bool = None,
        pwr3: bool = None,
        pwr4: bool = None,
    ) -> dict:
        """Set the power state of the device."""
        state = {}
        if did is not None:
            state["did"] = did
        if pwr1 is not None:
            state["pwr1"] = int(bool(pwr1))
            state["scenarioswitch_1"] = int(bool(pwr1))
        if pwr2 is not None:
            state["pwr2"] = int(bool(pwr2))
            state["scenarioswitch_2"] = int(bool(pwr2))
        if pwr3 is not None:
            state["pwr3"] = int(bool(pwr3))
            state["scenarioswitch_3"] = int(bool(pwr3))
        if pwr4 is not None:
            state["scenarioswitch_4"] = int(bool(pwr4))

        packet = self._encode(2, state)
        response = self.send_packet(0x6A, packet)
        e.check_error(response[0x22:0x24])
        return self._decode(response)
```

dj-fiorex, 2022-02-10:
just tried it but nothing changed, same error:
```
  File "/test/broadlink/device.py", line 309, in send_packet
    raise e.NetworkTimeoutError(
broadlink.exceptions.NetworkTimeoutError: [Errno -4000] Network timeout: No response received within 10s
```

dj-fiorex, 2022-02-10:
Now a GET request to homeassistant.local:5000/00000000000000000000ec0bae371ee1 give me this json response:
`{ "status": 0 }`
same for GET on homeassistant.local:5000/00000000000000000000ec0bae371ee1/1

stevendodd, 2022-02-10:
Can you reproduce the same failure; the fact that it keeps state means it acts like a switch. 

The only way to further is to listen to packets from the hub to/from your phone [ ](https://play.google.com/store/apps/details?id=app.greyshirts.sslcapture&hl=en_GB&gl=US)

dj-fiorex, 2022-02-10:
no i can't have a notification without setting an app notification

stevendodd, 2022-02-10:
Well there's not much more I can do at the moment I'm sorry, I am fairly sure that your button does not update the hub and it is in fact your app that is polling the button via the hub somehow.  The only way to figure it out will be to intercept network traffic as you have been doing and then decoding the packets as per the protocol specified in this repository.  I have looked through the packet captures that allowed me to reverse engineer the LC one switch and all of the traffic between the phone and hub was via UDP; the only TCP traffic was between the application and the broadlink servers

stevendodd, 2022-02-10:
If you look above those 256 B UDP responses was each of the sub devices getting queried from my application via the hub. It does it constantly in the background

dj-fiorex, 2022-02-10:
just tried with your app, here it is!
[image]
[image]
[image]
[image]

This is the packets sniffed when i click the button!

stevendodd, 2022-02-10:
Should be able to find the Mac addresses of the hub in reverse order in those hex dumps fairly easily

dj-fiorex, 2022-02-10:
Please can you help me? I have never done anything like this

stevendodd, 2022-02-10:
Yeah I'll give it a go tomorrow and see if I can decode the first request / response

stevendodd, 2022-02-10:
In the broadlink app when you click on device in play could you tell me what device has the Mac address  EC b0 ae 38 FC 83

stevendodd, 2022-02-11:
> Now a GET request to homeassistant.local:5000/00000000000000000000ec0bae371ee1 give me this json response: `{ "status": 0 }` same for GET on homeassistant.local:5000/00000000000000000000ec0bae371ee1/1
> 
> i tried clicked a button and the response change back to the "original" one:
> 
> ```
> {
> "datatype": "fw_snapshot_v1",
> "heartbeat": 0,
> "lb_online1": 1,
> "lowbattery": 0,
> "scenarioswitch_1": 1,
> "scenarioswitch_2": 0,
> "scenarioswitch_3": 0,
> "scenarioswitch_4": 0
> }
> ```
> 
> I want to emphasize that the sr3-4key is not a switch, it is a portable button, so it is no sense to set state on a stateless button, we need, in my opinion, focus on the hub that receive the signal from the button

It looks like just from the traffic pattern above your app is broadcasting without a specific address and then your hub is responding to the broadcast when the button is pressed. Looking at your quote above perhaps we also need to set the heartbeat parameter; but first I would love you to try:

1. Press button one
2. Connect to and authenticate to your hub via the python library
3. Run the get state method
4. turn on packet capture and then run

```
set_state(did,None,1,None,None)
```
5. Run get state method again
6. Paste python output here for all of the above
7.

stevendodd, 2022-02-11:
Thank you for doing that, I will look at decoding those packets for you

dj-fiorex, 2022-02-11:
if you can link me a guide i will try too, 4 eyes is better than 2 👍

dj-fiorex, 2022-02-11:
yes, i saw protocol.md, but what type of messages could that be? command type?

stevendodd, 2022-02-11:
My working theory is that the hub send a broadcast request and the button responds like what you captured above. The hub then potentially updates the broadlink servers directly and executes any routines that have been configured.

Out of interest could you please let me know what your intention is in terms of using the library? Do you want to poll the button for a button press?

All of the code I have written so far is command type however as we have tested and seen above that doesn't work and so I think it will be a discovery/broadcast message with a response from the button potentially via the hub

stevendodd, 2022-02-11:
If you look in device.py it does a broadcast message when trying to discover devices on the network

stevendodd, 2022-02-11:
Code in device.py should work for the listener almost out of the box. If you run

```
devices = broadlink.discover()
```

And while you are discovering press a button could you please print the response. Could you try doing it a couple of times using different buttons

stevendodd, 2022-02-11:
```
def scan(
    timeout: int = DEFAULT_TIMEOUT,
    local_ip_address: str = None,
    discover_ip_address: str = DEFAULT_BCAST_ADDR,
    discover_ip_port: int = DEFAULT_PORT,
) -> t.Generator[HelloResponse, None, None]:
    """Broadcast a hello message and yield responses."""
    conn = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    conn.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    conn.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)

    if local_ip_address:
        conn.bind((local_ip_address, 0))
        port = conn.getsockname()[1]
    else:
        local_ip_address = "0.0.0.0"
        port = 0

    packet = bytearray(0x30)
    packet[0x08:0x14] = Datetime.pack(Datetime.now())
    packet[0x18:0x1C] = socket.inet_aton(local_ip_address)[::-1]
    packet[0x1C:0x1E] = port.to_bytes(2, "little")
    packet[0x26] = 6

    checksum = sum(packet, 0xBEAF) & 0xFFFF
    packet[0x20:0x22] = checksum.to_bytes(2, "little")

    start_time = time.time()
    discovered = []

    try:
        while (time.time() - start_time) < timeout:
            time_left = timeout - (time.time() - start_time)
            conn.settimeout(min(DEFAULT_RETRY_INTVL, time_left))
            conn.sendto(packet, (discover_ip_address, discover_ip_port))

            while True:
                try:
                    resp, host = conn.recvfrom(1024)
                except socket.timeout:
                    break

                devtype = resp[0x34] | resp[0x35] << 8
                mac = resp[0x3A:0x40][::-1]

                if (host, mac, devtype) in discovered:
                    continue
                discovered.append((host, mac, devtype))

                name = resp[0x40:].split(b"\x00")[0].decode()
                is_locked = bool(resp[0x7F])
                yield devtype, host, mac, name, is_locked
    finally:
        conn.close()
```

It will print that it's found the S3 hub but it's the payload that is interesting - In the case of the response you received above `34 32 35 37 33` which is decoded into the variable name above perhaps you could print that out

dj-fiorex, 2022-02-11:
Ok, i changed scan function and i tried your code:
```
import broadlink

buttonsDid = "00000000000000000000ec0bae371ee1"

def main():
    print("Hello World!")
    devices = broadlink.discover()
    print(devices)

if __name__ == "__main__":
    main()

```
this is what it print:
```
[broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False)]
```

stevendodd, 2022-02-11:
Did you allow multiple entries based on mine as well? Did you press a couple of buttons during discovery?

dj-fiorex, 2022-02-11:
what you mean with "Did you allow multiple entries based on name as well?"
Yes i clicked all buttons, and when i clicked the button 1 i received the notification on the phone.

this is packet.hex()
`000000000000000001000000e607281316050b0200000000000000000000000006c00000000006000000000000000000`
this is packet
`bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\xe6\x07)\x13\x16\x05\x0b\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\xc0\x00\x00\x00\x00\x06\x00\x00\x00\x00\x00\x00\x00\x00\x00')
`

stevendodd, 2022-02-11:
```
if (host, mac, devtype) in discovered:
                    continue
discovered.append((host, mac, devtype))
```

Will discard multiple entries even if the name is different.

I think we might need to take a step back. Could you try the packet capture again and see if there is a difference in the response when a different button is pressed

stevendodd, 2022-02-11:
Look at the packet.hex and compare it to the request from your number one capture above

dj-fiorex, 2022-02-11:
oh sure, this is the number 1 image export file: https://we.tl/t-VQyDta74lu

stevendodd, 2022-02-11:
Something went wrong, you can just copy and paste the hex strings using normal copy and paste,

dj-fiorex, 2022-02-11:
> ```
> if (host, mac, devtype) in discovered:
>                     continue
> discovered.append((host, mac, devtype))
> ```
>
> Will discard multiple entries even if the name is different.
>
> I think we might [...]

Wait, wait, wait, i removed the code that skip if a device with the same name already exist and boom:
```
[broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False), broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False), broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False), broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False), broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False), broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False), broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False), broadlink.hub.s3(('192.168.1.74', 80), mac=b'\xec\x0b\xae8\xfc\x83', devtype=42573, timeout=10, name='42573', model='S3', manufacturer='Broadlink', is_locked=False)]
```
I know i clicked too many times lol

dj-fiorex, 2022-02-11:
> 

```
5a a5 aa 55 5a a5 aa 55  01 00 00 00 e6 07 28 16  Z..UZ..U ......(
14 04 0a 02 00 00 00 00  01 08 00 0a 3a a1 00 00  ....... ....:...
ef c4 00 00 00 00 06 00  00 00 00 00 00 00 00 00  ........ ........

5a a5 aa 55 5a a5 aa 55  01 00 00 00 e6 07 28 16  Z..UZ..U ......(
14 04 0a 02 00 00 00 00  01 08 00 0a 3a a1 00 00  ....... ....:...
05 cd 00 00 00 00 07 00  00 00 00 00 00 00 00 00  ........ ........
4d 59 0e 58 4d a6 4a 01  a8 c0 83 fc 38 ae 0b ec  MY.XM.J. ....8...
34 32 35 37 33 00 00 00  00 00 00 00 00 00 00 00  42573... ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 02 00  ........ ........

5a a5 aa 55 5a a5 aa 55  01 00 00 00 e6 07 29 16  Z..UZ..U ......)
14 04 0a 02 00 00 00 00  01 08 00 0a 3a a1 00 00  ....... ....:...
f0 c4 00 00 00 00 06 00  00 00 00 00 00 00 00 00  ........ ........

5a a5 aa 55 5a a5 aa 55  01 00 00 00 e6 07 29 16  Z..UZ..U ......)
14 04 0a 02 00 00 00 00  01 08 00 0a 3a a1 00 00  ....... ....:...
06 cd 00 00 00 00 07 00  00 00 00 00 00 00 00 00  ........ ........
4d 59 0e 58 4d a6 4a 01  a8 c0 83 fc 38 ae 0b ec  MY.XM.J. ....8...
34 32 35 37 33 00 00 00  00 00 00 00 00 00 00 00  42573... ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 02 00  ........ ........

5a a5 aa 55 5a a5 aa 55  01 00 00 00 e6 07 2a 16  Z..UZ..U ......*
14 04 0a 02 00 00 00 00  01 08 00 0a 3a a1 00 00  ....... ....:...
f1 c4 00 00 00 00 06 00  00 00 00 00 00 00 00 00  ........ ........

5a a5 aa 55 5a a5 aa 55  01 00 00 00 e6 07 2a 16  Z..UZ..U ......*
14 04 0a 02 00 00 00 00  01 08 00 0a 3a a1 00 00  ....... ....:...
07 cd 00 00 00 00 07 00  00 00 00 00 00 00 00 00  ........ ........
4d 59 0e 58 4d a6 4a 01  a8 c0 83 fc 38 ae 0b ec  MY.XM.J. ....8...
34 32 35 37 33 00 00 00  00 00 00 00 00 00 00 00  42573... ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
00 00 00 00 00 00 00 00  00 00 00 00 00 00 02 00  ........ ........

```

dj-fiorex, 2022-02-11:
No, I cheered for nothing, tried to run again the program, this time without click a button and the response is the same :(
So is the same packet

stevendodd, 2022-02-11:
Yes, can you go back to capturing packets and seeing if you can see a difference between one button press or another, you may need to go back to Wireshark although I think you should try the phone packet capture first

stevendodd, 2022-02-11:
```
000000000000000001000000e607281316050b0200000000000000000000000006c00000000006000000000000000000
5aa5aa555aa5aa5501000000e607281614040a02000000000108000a3aa10000efc40000000006000000000000000000
```

stevendodd, 2022-02-11:
Is the payload in the response: `0x3432353733`
[image]

dj-fiorex, 2022-02-11:
sorry for the late reply, I am continuing to test reading with packet capture, but I am unable to bind a package to the click of a button.
Very good job @stevendodd, so what does it mean? LOL

stevendodd, 2022-02-11:
It's the name of the S3 hub you printed above in ASCII

dj-fiorex, 2022-02-11:
I can't reproduce the same thing when a button is clicked, how can i export to another format?

stevendodd, 2022-02-11:
If the button click is not going to the phone/app then the only way we'll be to route traffic from the button via somewhere where you can intercept it to the hub

[42 more replies]

closed 2022-03-19 by felipediel