mjg59/python-broadlink #648: No encryption during provisioning?

Lull3n, 2022-01-18:
Was looking to implement the provisioning directly on android, and had a look at the setup method. Am i misunderstanding something here or are my wifi creds sent in over open wifi?

felipediel, 2022-01-23:
It's not open Wi-Fi, you need to connect to the Wi-Fi network that your Broadlink device generates in AP setup mode.

Lull3n, 2022-01-25:
Hmm, might be that I've misunderstood wifi principles here, but aren't WiFi's with no password unencrypted?

Lull3n, 2022-01-25:
I feel I'm probably misunderstanding here? What I mean is if the wifi is not protected, it's not encrypted? And if there is no application layer encryption anyone sniffing the radio will then be able to read my wifi creds? I had a look at the JS in the "wifi login page" generated by my bulb, and it applies what appears to (to my untrained JS eyes) application layer encryption, which seems to support my suspections somewhat.

felipediel, 2022-01-25:
It uses WPS, the communication is encrypted.

Lull3n, 2022-01-25:
I fear my  broadlink "Smart Bulb" doesent. There are no buttons on the bulb, no PIN requests, and netsh reports:
`    Network type           : Infrastructure
    Radio type             : 802.11n
    Authentication         : Open
    Cipher                 : None
`

felipediel, 2022-01-25:
Sometimes it's not a button, but a specific way you turn the power on and off to put it in AP setup mode. I don't have the bulb, I don't know how it works exactly, but the configuration is probably similar to the other devices with Broadlink DNA.

Lull3n, 2022-01-25:
Fair, i guess if there is WPS the device should only accept one concurrent connection? Otherwise the whole point of encrypting the data would be lost as anyone could just obtain the keys. I can connect multiple devices to the AP at the same time, and anyways as stated windows reports the connection as unencrypted. So, im getting fairly confident that the wifi creds are actually being broadcast unencrypted. Seing as broadlink has mitigated this with application level encryption i think you should look in to this, or at the very least disable the setup method for connections that have no encryption. I can take a swing at sniffing the packets if you need me to, and i can upload the JS encryption functions used by broadlink if you want to, but im afraid i can't make any PR's as my python is horrible.

felipediel, 2022-03-19:
When we connect via WPS, the Broadlink device restricts the connection to the client and the Wi-Fi configuration is done with encryption. WPS is like connecting with passwords, but you press a button to grant permission to the first device that connects.

Lull3n, 2022-03-19:
I'm well aware what wps is, as stated there is no WPS in my lb2 (can get you the hex code if you want) the encryption is instead done in the application layer using a rabbit cipher that's run in JS on the the phone/tablet used for provisioning. As I've also already stated windows also reports the connection as unencrypted hence any credentials sent on the network without adding application layer encryption are subject to sniffing, and the only security granted here is by obscurity.

Lull3n, 2022-03-19:
Really cool project, and super greatful for all the work ofc, but you should at least inform your users that the credentials are sent in cleartext instead of just saying there's WPS (which there at least isn't on my device) and closing the issue.

felipediel, 2022-03-19:
Okay, sorry, I'm short on time for the project so I end up having to make choices. I'll leave it open a little longer in case anyone wants to improve this with a PR.

felipediel, 2022-03-19:
If you could paste that Js here, it would be of great help.

Lull3n, 2022-03-21:
No worries mate! Sure! The bulbs i have are LB27-R1 mentioned in #522 sorry about confusing it with LB2. When provisioning the bulb send a page with some gnarly ass JS. I guess the actual encryption takes place in "_rabbit" but there's also a lot of stuff generating keys and such.

`var cou = "Deutschland",
  la = "en-us",
  n = "001",
  hd = "device-heartbeat-deu-6dc239d5.ibroadlink.com",
  gd = "device-gateway-deu-6dc239d5.ibroadlink.com",
  rg = "Europe",
  ac = "Europe",
  r_f_ = 1,
  lan = [
    "Select Wi-Fi network",
    "Input Wi-Fi password",
    "OK",
    "Input ",
    " ",
    "phone",
    "",
    "Close",
    "email",
    "",
    "Format error",
    "SMS",
    "Select the router to which the device will be connected (only 2.4GHz)",
    "",
    " to the router",
    "1. After the device is online, a confirmation [s] will be sent to {s}.",
    "3. If you do not receive the email in 5min, please check your span in email. ",
    "Tap ''Connect now'' to connect the device",
    "The device is connecting ",
    "Device configuration timed out. Please make sure the information filled in is correct.",
    "OK",
    "2. Search and enable “BroadLink” skill/service in Alexa or Google Home app, then sign in with [s] use the skill. ",
    "Please close this page if you have received the confirmation [s].",
    "",
    "Internet connection issue on router.",
    "Back",
    "Connect now",
    "I have read the above information",
    "",
    "",
    "Use [s] to activate your device",
    "Input your [s] to sign up",
    "",
    "",
    "",
    "",
    "",
    "OK",
    "",
    "V1.3.3",
    "This page might be closed automatically after the device is successfully connected to the router.",
    "Next",
    "Loading",
    "email",
    "1. After the device is online, a confirmation email will be sent to %1s.",
    "2. Search and enable ''BroadLink'' skill/service in Alexa or Google Home app, then sign in with %1s use the skill.",
    "3. If you do not receive the email in 5min, please check your spam in email.",
    "Sign in with [s] if you already have BroadLink account",
    "",
    "",
    "What is name of your device?",
    "Enter the name of your device",
    "Enter the [s] name",
    "The name cannot contain special characters",
    "Device name exists",
    "4. If you still cannot find the email in your spam folder, that means your device is failed to connect to Internet. Please check your network connection and Wi-Fi password. Reset the device following instructions in the product manual and try again.",
  ];
var d_nam = "Color Bulb";
var vGroups = [];
[... 1008 lines cut ...]
        g[2] + ((g[1] << 16) | (g[1] >>> 16)) + ((g[0] << 16) | (g[0] >>> 16));
      x[3] = g[3] + ((g[2] << 8) | (g[2] >>> 24)) + g[1];
      x[4] =
        g[4] + ((g[3] << 16) | (g[3] >>> 16)) + ((g[2] << 16) | (g[2] >>> 16));
      x[5] = g[5] + ((g[4] << 8) | (g[4] >>> 24)) + g[3];
      x[6] =
        g[6] + ((g[5] << 16) | (g[5] >>> 16)) + ((g[4] << 16) | (g[4] >>> 16));
      x[7] = g[7] + ((g[6] << 8) | (g[6] >>> 24)) + g[5];
    },
  };
  blrabbit.encrypt_ = function (msg, pwd) {
    return Rabbit.encrypt(msg, pwd);
  };
  blrabbit.decrypt_ = function (msg_ed, pwd) {
    return Rabbit.decrypt(msg_ed, pwd);
  };
  blrabbit.base64Encode = function (message) {
    return Rabbit.base64Encode(message);
  };
})(
  typeof module !== "undefined" && module.exports
    ? module.exports
    : (self.blrabbit = self.blrabbit || {})
);
`

Lull3n, 2024-06-12:
I deleted the JS, as i realized there might be some stuff in it pointing to my device. Also I would like to apologize for my tone in the previous communications.

closed 2024-06-12 by Lull3n