mjg59/python-broadlink #650: Can't get the learning code from IR/RF.

luvmex, 2022-01-24:
---Basic info--
Device: Broadlink RM Pro 4
Wifi: 2.4 (excellent)
IP: Get
Version: Updated
Device Lock: Off
---------------

Just trying to learn the code by typing: "device.enter_learning()"
On-time, the device lights up, and I trigger the remote control(Press multiple times). But nothing happened.

I'm not sure it's getting the code or not, so I typing: "packet = device.check_date()"

And I get the result as below:
--
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/broadlink/remote.py", line 37, in check_data
    return self._send(0x4)
  File "/home/pi/.local/lib/python3.9/site-packages/broadlink/remote.py", line 82, in _send
    e.check_error(resp[0x22:0x24])
  File "/home/pi/.local/lib/python3.9/site-packages/broadlink/exceptions.py", line 152, in check_error
    raise exception(error_code)
broadlink.exceptions.StorageError: [Errno -5] The device storage is full
--

Have no idea where and what goes wrong with the setting part, Hope to have any solution, and thanks in advance.

felipediel, 2022-01-25:
Hi. Make sure you are using the right class to control the device. You can also select the class automatically with:
```python3
import broadlink as blk
device = blk.hello("192.168.0.16")  # Example IP address
```

Amoenus, 2022-03-07:
I have the same problem with my 智能遥控 Broadlink RM4 pro 0x61a2

felipediel, 2022-03-07:
Hi. Same solution

closed 2024-04-21 by felipediel