mjg59/python-broadlink #656: Exiting learning_mode/learing_mode timeout for remote devices

zroger49, 2022-02-05:
Hello! 
First of all, thanks for the great integration, the code seems very well written and its super easy to use . 

I would like to known if there is any way for my universal remote to exit learning mode. 

For my use case I'm developing an application in javascript which interacts with the broadlink API to learn and send commands from my application. I have a visual indication for when the device is in learning mode (on my application), but I would like my rm4pro to also match this. I guess this would be more like a feature request. Let me known if I can help with anything

felipediel, 2022-02-06:
Hi. Thank you!
Some devices support device.cancel_sweep_frequency() to exit learning.

zroger49, 2022-02-06:
Perfect, this works for my needs

closed 2022-02-06 by zroger49