mjg59/python-broadlink #657: [Question] Transforming IR codes in the python-broadlink format

helen-fornazier, 2022-02-05:
Hi, first of all, thanks for this project, really useful.

I was reading [protocol.md](https://github.com/mjg59/python-broadlink/blob/master/protocol.md) to try to figure out the package formats, and I'm trying to understand it.

I know that the following payload turns on a Samsung TV:

```
020026008c0092961039103a1039101510151014101510151039103a10391015101411141015101510141139101510141114101510151014103a10141139103911391037123a10391000060092961039103911391014111410151015101411391039103a101411141015101510141015103911141015101510141015101510391015103911391039103a1039103911000d05000000000000000000000000
```

Where:
`0200` is fixed by the protocol
`26` means IR
`008c` is the size of the following data, which is 140
`...` then it comes pulse lengths that I didn't understand the units, could anyone help me?
`00d05` means the end of the message
`000000000000000000000000` trailing zeros for silence period

I see the example of the Optoma projector in the docs, but I didn't really understand. I thought  that the projector started with a ON pulse of 8920 µs and OFF pulse of 4450 µs,  so ` 8920 * 269 / 8192 = 0x124`  and `4450 * 269 / 8192 = 0x92`, but the docs says that the data starts with `0x00 0x1 0x24 0x92`, why not `0x01` ? In the example it uses 3 bytes for ON and one byte for OFF, does this repeat for the following pulses? How do I know how to read the bytes?

How do I interpret the code above? Considering the first block `92961039`, what does this mean?

Thank you a lot!

---

My end goal is: I would like get ID codes from IR data bases around there ( [like this one for example](http://www.remotecentral.com/cgi-bin/codes/samsung/tv_functions/) ) , and without doing learning, send out this IR commands.

I understand that `python-broadlink` uses pulse length duration, right? So I would like to better understand the payload, so I can write a tool capable of transforming IR Codes from the data base into a payload for this library. (Note: I intent to make this tool available so other people can use too).

felipediel, 2022-02-06:
Hi! Thank you!
You may want to have a look at https://github.com/elupus/irgen and https://github.com/marcan/circa. These projects are more focused on encoding/decoding Broadlink IR packets.

helen-fornazier, 2022-02-15:
Thanks a lot!

I also found this one: https://github.com/haimkastner/broadlink-ir-converter

So I converted the power on TV, without the `0200` prefix in `node` console

```
> b = require('broadlink-ir-converter')
> payload = '26008c0092961039103a1039101510151014101510151039103a10391015101411141015101510141139101510141114101510151014103a10141139103911391037123a10391000060092961039103911391014111410151015101411391039103a101411141015101510141015103911141015101510141015101510391015103911391039103a1039103911000d05000000000000000000000000'
> raw = b.broadlinkToPulesArray(payload);
> console.log(JSON.stringify(raw))
[0,4446,4568,487,1735,487,1766,487,1735,487,639,487,639,487,609,487,639,487,639,487,1735,487,1766,487,1735,487,639,487,609,517,609,487,639,487,639,487,609,517,1735,487,639,487,609,517,609,487,639,487,639,487,609,487,1766,487,609,517,1735,487,1735,517,1735,487,1674,548,1766,487,1735,487,46776,4446,4568,487,1735,487,1735,517,1735,487,609,517,609,487,639,487,639,487,609,517,1735,487,1735,487,1766,487,609,517,609,487,639,487,639,487,609,487,639,487,1735,517,609,487,639,487,639,487,609,487,639,487,639,487,1735,487,639,487,1735,517,1735,487,1735,487,1766,487,1735,487,1735,517]
undefined
```

(and this is correct,  since I tested by sending raw commands from esp32)

So if I take the two first non-zero pulses of `4446` and `4568`:

```python
>>> hex(round(4446*269/8192))
'0x92'
>>> hex(round(4568*269/8192))
'0x96'
>>> 
```

Which matches with the `9296` part of the payload.

And now I understand the Optoma example,  because in the docs, it says:
`Each value is represented by one byte. If the length exceeds one byte then it is stored big endian with a leading 0.`
I had missed this part, this is why it starts with `0x00 0x1` and not `0x1 0x24`.

closed 2022-02-15 by helen-fornazier