mjg59/python-broadlink #659: Question about next update for adding new light bulb codes

jr3us, 2022-02-11:
Greetings!

I wonder if the library will be updated soon as I have 4  LB26 lightbulbs that really want to play in the home assistant world, and i notice that the issue for adding the code has been closed and was added to the 'add new codes' version of the repository in october, but hasn't made it to release yet.

I do have an issue set up in the home assistant github repo and i notice there has been some activity on it.

I'm trying to be real patient and not be a pain. :-)

feel free to close this issue or I can.

Thank you much for all your effort in creating this tool/library!

jr3us, 2022-02-19:
I'm closing this as the changes have been migrated to the home-assistant dev instance.

jr3us, 2022-03-04:
@felipediel , I have done the docker and pip3 command to install the new_product_ids, and I am getting an error [Errno -6] Structure is abnormal within home assistant.   I then loaded the new product ids via pip on another machine, and am getting this error there is well.

Is there any information i can provide to help? These are the same bulbs I referenced in the home assistant issue https://github.com/home-assistant/core/issues/61911

Regards!

felipediel, 2022-03-04:
What is your pid? 0x644C?

jr3us, 2022-03-04:
the pid is 644E.

felipediel, 2022-03-04:
Ok, could you please check if it works with the LB1 class? If you don't know how to do it I can create a branch for you later, I'm at work now.

jr3us, 2022-03-04:
I'm not sure how to ... Do I just need to move the LB26 line with 644E to the LB1 section?

jr3us, 2022-03-04:
in  __init__.py ?

felipediel, 2022-03-04:
Yes, then try to control it with `device.auth()` followed by `device.get_state()`.

jr3us, 2022-03-04:
ok, it worked, and i am getting valid info in get_state with it under lb1 instead of lb2

felipediel, 2022-03-04:
Thank you! I will fix it.

jr3us, 2022-03-04:
oh by the way, I was chatting with the experts on discord, and they indicate the following regarding testing/customizing for modules such as this broadlink one. I am quoting here from the discord channel:
ah, if you're trying to customize an integration this is not the way
you would want to make use of the custom_integrations directory, basically creating a copy of the core integration there
integrations channel for more help with that
jr3us — Today at 10:52 AM
so custom_integrations directory would over ride whatever version is installed via the integrations page?
cogneato — Today at 11:07 AM
yeah
This can be done manually, but it is also where HACs downloads and places all custom integrations. Before HACs existed, it was all done by hand and following the docs for all the unofficial integrations

felipediel, 2022-03-04:
It depends on the problem you are solving. I'm not sure you can update the library via custom_integrations folder. It was not possible in the past at least. If you manage to do it, please let me know.

jr3us, 2022-03-04:
Yeah, I'm not sure you could use pip to install it... it would have to be dumped as a zip somehow maybe... waiting on an answer from 'cogneato' on discord regarding pip.

closed 2022-03-05 by felipediel