mjg59/python-broadlink #665: RF Packet Captured- Please Help

migracing, 2022-03-19:
I ran everything and I captured the RF packet for a ceiling fan I'm trying to control. I even tested it to make sure the packet I captured worked. From the command line I can control my fan perfectly fine. I need help converting the packet to base64 so that I can use it with homebridge or home assistant. 

The following is what I get after capturing the packet. I have no idea what to do with that.

>>> packet = device.check_data()
>>> print(packet)
b'v\xc0\xb0\x04\x90\xca\x04\x00\x0c\x1e\x0e\x1f\r\x1e\x0e\x10\x1cK\n>\x1e\x10\x1c\x1f\r\x10\x1d\x1e\r\x10\x0f+\x0f\x1f\r\x1f\r\x10\x1c\x10\x1cK\x0e\x10\x1c\x1f\r\x10\x1d\x1e\x0e\x00\x01\xb9\x0f\x1e\r\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x1e\x0e\x1e\x0e\x10\x1d\x0f\x1d\x10\x1c\x1f\r\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x00\x01\xb9\x0e\x1f\r\x1f\r\x1f\x0e\x0f\x1d\x0f\x1d\x1f\x0e\x1e\x0e\x0f\x1d\x10\x1c\x1f\r\x10\x1d\x1e\x0e\x10\x1c\x1f\r\x1f\r\x1f\x0e\x0f\x1d\x10\x1c\x10\x1d\x1e\x0e\x10\x1c\x1f\r\x10\x1d\x1e\x0e\x00\x01\xba\x0e\x1f\r\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x10\x1c\x1f\x0e\x0f\x1d\x1f\r\x1f\r\x1f\r\x10\x1d\x10\x1c\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x10\x1d\x1e\x0e\x00\x01\xba\x0e\x1e\x0e\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x1f\r\x1f\r\x10\x1d\x10\x1c\x10\x1d\x1e\x0e\x10\x1c\x1f\x0e\x0f\x1d\x1f\r\x00\x01\xbb\x0e\x1e\x0e\x1e\x0e\x1e\x0e\x0f\x1d\x10\x1c\x1f\r\x1f\x0e\x0f\x1d\x10\x1d\x1e\x0e\x0f\x19"\x0e\x10\x1d\x1e\x0e\x1f\r\x1e\x0e\x10\x1c\x10\x1d\x10\x1c!\x0b\x10\x1d\x1f\r\x0f\x1d\x1f\x0e\x00\x01\xba\x0e\x1e\x0e\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x1f\r\x10\x1d\x10\x1c\x1f\r\x10\x1d\x1e\x0e\x10\x1c\x1f\r\x1f\r\x1f\x0e\x0f\x1d\x10\x1c\x10\x1d\x1f\r\x10\x1c \r\x0f\x1d\x1f\r\x00\x01\xbb\x0e\x1e\x0e\x1e\x0e\x1e\x0e\x10\x1c\x10\x1d\x1f\r\x1f\r\x10\x1c\x10\x1d\x1f\r\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x1f\r\x1f\x0e\x0f\x1d\x10\x1c\x10\x1d\x1f\r\x10\x1c\x1f\r\x10\x1d\x1f\r\x00\x01\xbb\x0e\x1e\x0e\x1e\x0e\x1e\x0e\x10\x1c\x10\x1d\x1e\x0e\x1f\r\x10\x1c\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x0f\x18#\x0e\x1f\r\x1f\x0e\x0f\x1d\x10\x1c\x10\x1d\x1e\x0e\x10\x1c\x1f\x0e\x0f\x1d\x1f\r\x00\x01\xbb\x0e\x1e\x0e\x1f\r\x1f\r\x10\x1c\x10\x1d\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x10\x1c\x1f\x0e\x0f\x1d\x1f\r\x1f\r\x1f\x0e\x0f\x1d\x10\x1c\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x0f\x1d\x1f\r\x00\x01\xbb\x0e\x1f\r\x1f\r\x1f\r\x10\x1c\x10\x1d\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1e\x0e\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x1f\x0e\x1e\x0e\x0f\x1d\x10\x1c\x11\x1c\x1f\r\x10\x1c\x1f\x0e\x0f\x1e\x1e\r\x00\x01\xbb\x0e\x1e\x0e\x1f\r\x1f\r\x10\x1d\x10\x1c\x1f\r\x1f\r\x10\x1d\x10\x1c\x1f\x0e\x0f\x1d\x1f\r\x10\x1c\x1f\x0e\x1f\r\x1f\r\x10\x1d\x0f\x1d\x10\x1c\x1f\r\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x00\x01\xbb\r\x1f\x0e\x1e\x0e\x1e\x0e\x0f\x1d\x10\x1c\x1f\x0e\x1f\r\x10\x1c\x10\x1d\x1e\x0e\x10\x1c\x1f\r\x10\x16\x05\x04\x1f\r\x1f\r\x1f\r\x10\x1d\x10\x1c\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x0f\x1d\x1f\r\x00\x01\xbb\x0e\x1e\x0e\x1f\r\x1f\r\x10\x1d\x10\x1c\x1e\x0e\x1f\r\x10\x1d\x10\x1c\x1f\r\x10\x1d\x1f\r\x10\x1d\x1e\x0e\x1e\x0e\x1f\r\x10\x1c\x10\x1d\x10\x1c\x1f\r\x10\x1d\x1f\r\x10\x1d\x1e\x0e\x00\x01\xbb\r\x1f\r\x1f\x0e\x1e\x0e\x0f\x1d\x10\x1d\x1e\x0e\x1e\x0e\x10\x1c\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x0f\x14\'\x0e\x1f\r\x1f\x0e\x0f\x1d\x0f\x1e\x0f\x1d\x1f\r\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x00\x01\xbb\x0e\x1f\r\x1f\r\x1f\r\x10\x1d\x0f\x1d\x1f\r\x1f\x0e\x0f\x1d\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x10\x1c\x1f\x0e\x1f\r\x1f\r\x10\x1c\x10\x1d\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x10\x1d\x1e\x0e\x00\x01\xbb\x0e\x1e\x0e\x1e\x0e\x1f\r\x10\x1c\x10\x1d\x1e\x0e\x1f\r\x10\x1d\x0f\x1d\x1f\r\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x1e\x0e\x1f\r\x10\x1c\x10\x1d\x10\x1c\x1f\r\x10\x1d\x1f\r\x10\x1d\x1e\x0e\x00\x01\xbb\x0e\x1e\x0e\x1e\x0e\x1f\r\x10\x1c\x10\x1d\x1e\x0e\x1e\x0e\x10\x1c\x10\x1d\x1f\r\x10\x1d\x1e\x0e\x10\x1c\x1f\x0e\x1e\x0e\x1e\x0e\x0f\x1d\x10\x1d\x10\x1b \r\x10\x1d\x1f\r\x10\x1d\x1e\x0e\x00\x01\xba\x0e\x1f\x0e\x1e\x0e\x1e\x0e\x0f\x1d\x10\x1d\x1e\x0e\x1e\x0e\x10\x1c\x10\x1d\x1f\r\x0f\x1e\x1f\x0c\x11\x1c\x1f\r\x1f\x0e\x1e\x0e\x0f\x1d\x10\x1c\x10\x1d\x1f\r\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x00\x01\xba\x0e\x1f\r\x1f\x0e\x1e\x0e\x0f\x1d\x10\x1c\x1f\x0e\x1f\r\x10\x1c\x10\x1d\x1e\x0e\x10\x1d\x1e\x0e\x0f\x1d\x1f\x0e\x1e\x0e\x1e\x0e\x0f\x1d\x10\x1c\x10\x1d\x1f\r\x10\x1d\x1e\x0e\x10\x1c\x1f\r\x00\x01\xbc\r\x1f\r\x1f\x0e\x1e\x0e\x0f\x1d\x10\x1d\x1e\x0e\x1f\r\x10\x1c\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x10\x1c\x1f\r\x1f\x0e\x1e\x0e\x0f\x1d\x10\x1c\x10\x1d\x1f\r\x10\x1d\x1f\r\x10\x1c\x1f\x0e\x00\x01\xbb\r\x1f\x0e\x1f\r\x1e\x0e\x0f\x1d\x10\x1d\x1e\x0e\x1f\r\x10\x1c\x11\x1c\x1f\r\x10\x1d\x1e\x0e\x0f\x1d\x1f\r\x1f\x0e\x1e\x0e\x10\x1c\x10\x1d\x10\x1c\x1f\r\x10\x1d\x1e\x0e\x10\x1d\x1e\x0e\x00\x01\xbb\x0e\x1e\x0e\x1f\r\x1f\r\x10\x1c\x10\x1d\x1f\x0e\x1e\r\x10\x1d\x10\x1c\x1f\x0e\x0f\x1d\x1f\r\x10\x1d\x1e\x0e\x1e\x0e\x1f\r\x10\x1d\x0f\x1d\x10\x1d\x1e\x0e\x10\x1c\x1f\r\x10\x1d\x1f\r\x00\x01\xbb\x0e\x1e\x0e\x1e'

Any tips will be greatly appreciated.
Thank you.

kkernan, 2022-05-30:
Did you figure this out? I'm in the same place and wondering what the solution is.

migracing, 2022-05-30:
I did figure it out. 
I'm assuming you've already downloaded the "python-broadlink" files to your computer. Also I ran this on both Mac and Linux.

First thing we'll need is to obtain your broadlink device's: "type" and "mac address"
1. open python3
2. import broadlink
3. device = broadlink.hello('192.168.0.16')       # IP address of your Broadlink device.
4. print(device)       # this should print out your broadlink's info 
5. Note the type: (ex. "0x2787")
6. Note the mac address:  (ex. "35:EA:35:B4:5D:2C")

With that info you can exit python.

Now from terminal, we'll be executing a file from the "python-broadlink" project. 
We're looking for the "broadlink_cli" file.
I downloaded and unzipped the files to my desktop so I ran the following commands:

`cd Desktop/python-broadlink-master/python-broadlink-master/cli`

Substitute the following with your device type, ip address, and mac address (remove the colons),
We obtained this info above.

`./broadlink_cli --rfscanlearn --device "0x2787 192.168.0.16 35EA35B45D2C"`
  

After that your set, just follow the prompts.

I did notice that for RF frequencies it says to press and hold the button on the first step, but that didn't work for me, I had to repeatedly pressed the button for it to catch the sweeping frequency. 

*** if you want to learn IR codes just substitute the --rfscanlearn with --learn ***

That should do it. If you need any more tips let me know.

closed 2022-05-30 by migracing

kkernan, 2022-06-01:
Thanks. I have the packet info and if I send it from broadlink_cli the device in question responds appropriately, how do you convert that to something you can put into the homebridge configuration?