mjg59/python-broadlink #674: Device has no auth attribute?

Jibun-no-Kage, 2022-04-10:
Installed via 
# pip3 install 
# python3
>>> import broadlink
>>> broadlink.discover()
[broadlink.remote.rm4mini(('1.1.1.1', 80), mac=b'\xa0C\xb01\xf8>', devtype=25741, timeout=10, name='智能遥控', model='RM4 mini', manufacturer='Broadlink', is_locked=True)]
>>> theDevice=broadlink.discover()
>>> theDevice.auth()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'list' object has no attribute 'auth'

What is the issue?  Should have an object not a list, right?

Jibun-no-Kage, 2022-04-10:
And... the included example fails as well...

\# python3 broadlink_discovery
Discovering...
Traceback (most recent call last):
  File "/root/python-broadlink/cli/broadlink_discovery", line 17, in <module>
    if device.auth():
  File "/usr/local/lib/python3.9/dist-packages/broadlink/device.py", line 185, in auth
    e.check_error(response[0x22:0x24])
  File "/usr/local/lib/python3.9/dist-packages/broadlink/exceptions.py", line 152, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthenticationError: [Errno -1] Authentication failed

felipediel, 2022-05-23:
Hi, discover() returns a list, you need to use the index eg. theDevices[0]. The second problem is that the device is locked, you need to open the official app and unlock it in order to authenticate.

closed 2022-05-23 by felipediel