mjg59/python-broadlink #678: Broadlink device returns incorrect room_temp value (rounded, not floating)

rjulius23, 2022-04-25:
Hi everyone,

I have a Thermostat that is recognized by the Broadlink module as a Hysen thermostat. Everything works except that the room_temp is rounded to .0 or .5. So when the device shows 23.4 degrees on its screen, the module returns 23.0, and when it is 23.6 it returns 23.5. The reason for this is because the broadlink logic returns the double of the values as far as I know, so when it is 23 degrees it returns 46 and the code in the climate.py divides it by 2.0 hence the roiunding to 23.0 and 23.5 for example.

I would need some help to understand the payload format coming from the Broadlink device, but my understanding, that we may cut the missing .X value somehow. So when it is 23.4 it should return 46.8 but we get 46.0 only.

Here is a small snippet when the device actually showed **23.4 C**:
```
>>> devs[0].get_full_status()
**Payload: b'\x01\x03, -.\'\x10\x00*\x02#\x05\x00\x00\x00\x01"T\t\x1f\x1a\x02\x06\x00\x08\x00\x0b\x1e\x0c\x1e\x11\x00\x16\x00\x08\x00\x17\x00(\x1e\x1e\x1e,\x1e,\x1e'
Room temp: 46.0
Get Room temp Result: 23.0**
{'remote_lock': 0, 'power': 1, 'active': 0, 'temp_manual': 0, **'room_temp': 23.0**, 'thermostat_temp': 19.5, 'auto_mode': 0, 'loop_mode': 1, 'sensor': 0, 'osv': 42, 'dif': 2, 'svh': 35, 'svl': 5, 'room_temp_adj': 0.0, 'fre': 0, 'poweron': 1, 'unknown': 34, 'external_temp': 42.0, 'hour': 9, 'min': 31, 'sec': 26, 'dayofweek': 2, 'weekday': [{'start_hour': 6, 'start_minute': 0, 'temp': 20.0}, {'start_hour': 8, 'start_minute': 0, 'temp': 15.0}, {'start_hour': 11, 'start_minute': 30, 'temp': 15.0}, {'start_hour': 12, 'start_minute': 30, 'temp': 15.0}, {'start_hour': 17, 'start_minute': 0, 'temp': 22.0}, {'start_hour': 22, 'start_minute': 0, 'temp': 15.0}], 'weekend': [{'start_hour': 8, 'start_minute': 0, 'temp': 22.0}, {'start_hour': 23, 'start_minute': 0, 'temp': 15.0}]}
>>>
```

Can you please guide me towards the solution ? I can create the PR once I understand where is the issue.

fustom, 2022-10-02:
Hi @rjulius23,

I have the same issue, but I am pretty sure that the first part of the payload[17] 'unknown' contains the missing information for room_temp. (The second half maybe for the external temp?)

You should add the following decimal to your room_temp.
| Binary | Hex | Decimal |
| ------------- | ------------- | ------------- |
|1111| F |0|
|0000| 0 |1|
|0001| 1 |2|
|0010| 2 |3|
|0011| 3 |4|

In your case the 'unknown' is 34 hex. The first part is 3. That means you should add 0.4 to 23.0.

rjulius23, 2022-10-02:
Thanks. I will play around with it and open a PR.

fustom, 2022-10-02:
I think the easiest solution to get the correct room_temp is:
`payload[5] / 2.0 + ((((payload[17] >> 4) + 1) & 15) / 10.0)`

fustom, 2022-10-03:
@rjulius23 Can you create a PR? I don't need glory, just a (better) working code.

rjulius23, 2022-10-03:
I can just my todo list is pretty long nowadays but will try to take care
of it this week :)

closed 2024-04-21 by felipediel