mjg59/python-broadlink #679: Support for RM4-PRO - 0x520b

pereirapleandro, 2022-05-02:
When I try to configure my Broadlink RM4 PRO I receive an error `Unsupported device: 0x520b`.
Please, could you add this new version of Universal Remote RM4 PRO?
Thank you!

Mr-Beat, 2022-05-03:
I have the same problem: 
> 2022-05-03 11:42:23 ERROR (MainThread) [homeassistant.components.broadlink.config_flow] Unsupported device: 0x520b. If it worked before, please open an issue at https://github.com/home-assistant/core/issues

pereirapleandro, 2022-05-03:
The new RM4 Pro device was included on this PR https://github.com/mjg59/python-broadlink/pull/673 24 days ago.

Mr-Beat, 2022-05-05:
> +1 @felipediel ❤

felipecrs, 2022-05-07:
+1, I just bought two 0x520b and can't wait to integrate them in Home Assistant!

pereirapleandro, 2022-05-07:
@felipediel Could you help us, please? 🙏

NaturalSelectionPlayer, 2022-05-14:
Commit was created 1 month ago, but checks with current commit isnt pass well. And now we awaits correct code with last merge to master Br. There a lot of new devices IDs. Thx all.

eduardoll00, 2022-05-15:
+1 ! Waiting for the update

felipecrs, 2022-05-16:
@mjg59, @felipediel do you guys need any help? Like writing changelog, whatever? If so, let me know, I can try to help.

felipediel, 2022-05-23:
Hi guys, I just finished the update, it will be available in Home Assistant soon.

Thanks @felipecrs, this one is gone, but feel free to create a PR whenever you like :+1:

closed 2022-05-23 by felipediel