mjg59/python-broadlink #688: LB27 R1 0x644C - Home Assistant Issue

manunited10, 2022-05-26:
Hi.
I added some comments at the bottom of [this page](https://github.com/mjg59/python-broadlink/pull/644). It didn't get any attention so I created a new topic here.
There's a problem with my light which is recently added to this repo and Home Assistant. Honestly I'm not sure if it was a good idea to talk about here - but I did anyway! 
Thanks.

manunited10, 2022-06-07:
@felipediel any luck this could get an attention? 🤞

felipediel, 2022-06-07:
Hi, thanks for looking into this. I confirm the issue exists, but I'm short on time to work on it at the moment, if you can send a PR to Home Assistant with the changes it will be of great help

manunited10, 2022-06-08:
> Hi, thanks for looking into this. I confirm the issue exists, but I'm short on time to work on it at the moment, if you can send a PR to Home Assistant with the changes it will be of great help

I suspect it may fix my light but break others.

felipediel, 2022-06-08:
All lights have the same issue. There was a PR to fix them a while ago, but it was abandonned.

manunited10, 2022-06-09:
> All lights have the same issue. There was a PR to fix them a while ago, but it was abandoned.

I made a PR in homeassistant, but I made a bad mistake. Could you please delete it there? it's important. Thanks.

felipediel, 2022-06-21:
Sorry, there is no way to delete a PR. Why did you close it?

felipediel, 2024-04-17:
Fixed, thank you!

closed 2024-04-17 by felipediel