mjg59/python-broadlink #689: New device RM4C Mini with id 0x520d

titusthefox04, 2022-05-27:
Hi, I bought a device RM4C Mini on aliexpress, but I cannot integrate it into home assistant.
I receive the follow message from home assistant:

`Unsupported device: 0x520d. If it worked before, please open an issue at https://github.com/home-assistant/core/issues`

I installed python-broadlink and checked configuration using command ` devices = broadlink.discover()`
with the follow result:
`智能遥控 (0x520d / 192.168.1.57:80 / xx:xx:xx:xx:xx)`

device is a Broadlink RM4C mini

Can you add it to the library?

Thanks

titusthefox04, 2022-05-29:
I created a [pull request](https://github.com/mjg59/python-broadlink/pull/692) to solve this issue.

closed 2022-06-06 by felipediel

kevinkey619, 2022-07-12:
I'm on 2022.7.2 and it still says the RM4C Mini is an "unsupported device" when I attempt to add the integration.

titusthefox04, 2022-07-12:
I've checked the release notes and a version with device 0x520D was not built.