mjg59/python-broadlink #690: API can't communicate with device but app can (both in the same local network)

coupeborgward, 2022-05-28:
I add the new smart light bulb and the api can commincate with it without any issue. After I while I noticed that the light bulb is not a attached device on my router anymore but the broadlink app can still communicate/control it but the API can't since the IP address is not valid anymore. Has anyone seen this behaviour before.

coupeborgward, 2022-06-21:
I found the issue which unfortunately is a big one for this API. Broadlink are using smart hub technology which means one of the device in the network acts as a hub controlling the rest of the devices. The side affect of this is that not all devices will have an IP address assigned and cannot be communicate to by this API. Even assigning a static IP address etc will not help since the broadlink software itself is the controlling unit and not the router.

closed 2022-06-21 by coupeborgward

Lull3n, 2022-09-10:
Yeah, i just came across the same issue. Clever little fuckers... I think there ought to be a way to work around this, given that comms are conducted locally when the device is unlocked. Unfortunately this is way above my level, and i think the maintainers are understandibly pretty busy.

Lull3n, 2022-09-10:
A really stupid way to fix this (that seems to be working for now atleast), is to power on the lights at the same time. Seems they need some time before they start accepting connections.

MichielioZ, 2022-09-10:
I've had the same problem for a while now. I live in a small apartment and besides my RM4 Pro, I've got a smart bulb (LB27R1) which acts as a hub for my 2 smart plugs (SP4L-EU). I'm unable to use my 2 plugs in an automation like this and have reverted to some old RF plugs (having no feedback about their state is troublesome though).
Saw there is already a [pull request here](https://github.com/mjg59/python-broadlink/pull/669#issue-1174389815) by @felipediel 
Hope anybody reading this can help testing as I'm no good at decoding the message(s) :-(

coupeborgward, 2022-09-16:
I change over to meross smart light bulbs which work flawless. Their API communicated to the cloud which then establishes the comms to the light bulb itself. I have this setup running since 4 months now without any issues.