mjg59/python-broadlink #691: Question about firmware

AnilDaoud, 2022-05-29:
Is it possible to upgrade the firmware of rm mini 4 devices using this integration? Or must one use the official broadlink app/cloud and reset the device every time?

coupeborgward, 2022-05-29:
I would use the official app if possible

AnilDaoud, 2022-05-29:
This app is unfortunately broken for me at the moment. It asks for my phone to be connected to a 2.4Ghz wifi network, which it is, and the process to adding a new device never truly starts

Hence my search for an alternative

felipediel, 2022-06-21:
Not possible. If you're having trouble connecting to Wi-Fi, try changing your password to something simpler. See https://github.com/mjg59/python-broadlink/blob/master/TROUBLESHOOTING.md.

AnilDaoud, 2022-06-21:
thank you, wifi connectivity is more stable now for me, i will close this, even though i still wasn't able to upgrade firmware.

on a separate note, home assistant is not able to cope with changes in ip address (due to dhcp) of the broadlink device. i have put a static ip address for now and will make a feature request in home assistant.

coupeborgward, 2022-06-21:
Yep broadlink has implemented smart hub which means that one device will be acting as a hub controlling the rest of the devices in the network. This means that some devices will lose their IP address over time and cannot communicated to anymore. I had the same issue. Unfortunately this means that this API is not reliable anymore since not all device will have an IP address assigned to. I had this confirmed by broadlink developers. I'm using meross devices now which don't use smart hub feature