mjg59/python-broadlink #696: Is there a way to change the device name?  My RM4 Mini the name returned via this module is in chinese, but in the phone app in English?

Jibun-no-Kage, 2022-06-13:
Is there a way to change the device name?  My RM4 Mini the name returned via this module is in chinese, but in the phone app in English?

For example...
`        #
        broadlink.setup(SSID, PASSPHARSE, SECURITY)

        #
        DEVICES=broadlink.discover(discover_ip_address=BROADCAST)

        for theDevice in DEVICES:

                theAuthentication=theDevice.auth()
                theSensors=theDevice.check_sensors()

                #
                print(theDevice)
                print(theAuthentication)
                print(theSensors)

                #
                theTemperature,theHumidity=Values(theSensors, 'temperature','humidity')
                print(theTemperature)
                print(theHumidity)
`

The output...

`智能遥控 (Broadlink RM4 mini 0x648d / 192.168.1.XXX:80 / XX:XX:XX:XX:XX:XX)
True
{'temperature': 28.42, 'humidity': 23.39}
28.42
23.39
`

Really weird.  The above 智能遥控 translates to 'Smart remote control'.

felipediel, 2022-06-21:
Yes, you can call device.set_name("Something cool").

closed 2022-06-21 by felipediel

Jibun-no-Kage, 2022-06-21:
Thaanks!