mjg59/python-broadlink #698: Calling check_data instance error 'The device storage is full'?

Jibun-no-Kage, 2022-06-14:
Calling check_data instance error 'The device storage is full'?  No example of how to setup or configure storage?  I can discover devices, read sensors, etc. but after calling enter_learning, then check_data get above error?

Jibun-no-Kage, 2022-06-14:
Ah, found example of how to address the StorageError issue in the CLI script.  Not sure I like the idea that check_data throws and exception versus graceful continue until data arrives.  Maybe add a timeout parameter to check_data, so programmer does not have to create a timeout loop or such?

closed 2022-06-14 by Jibun-no-Kage

MitchTalmadge, 2024-10-01:
Had the same issue. Here's the CLI example that was referenced:

https://github.com/mjg59/python-broadlink/blob/b33bbdbbcf1270fb77f346f0e003273247d3ff2f/cli/broadlink_cli#L221-L232