mjg59/python-broadlink #699: Can Broad Link RM4 Mini (for example) Query device for power state?

Jibun-no-Kage, 2022-06-19:
Can Broad Link RM4 Mini (for example) Query device for power state?  For example with CEC, you can ask the device for its power state, on or off/standby.  But I have not seen any examples of doing the same via Broad Link devices?  Is such possible?

felipediel, 2022-06-21:
Not possible. If you have a Raspberry Pi, you can install Home Assistant and combine the HDMI-CEC integration with the Broadlink integration to achieve the result you want.

closed 2022-06-21 by felipediel

Jibun-no-Kage, 2022-06-21:
Humm... right now I am just using a simple ping to determine if the RM4 is on the network.  That works but is limited.

felipediel, 2022-06-21:
Ah, I understand now. I though you wanted to query the power state of the TV. A ping is ok to check the device, you could also use check_temperature() or any method that queries the state.